espressif/esp-azure #3: wifi_config need be static

sifourquier, 2018-07-26:
if wifi_config are no static he is deleted after initialise_wifi() and wifi do not connect

ustccw, 2018-08-01:
@sifourquier Hi, we have smart_config and airkiss to dynamic config your WiFi. you could refer to examples.

ustccw, 2018-08-27:
Closing due to inactivity. Please feel free to re-open or file a new issue if you have any more questions.

rtmistler, 2019-01-30:
I ran into this, on the iothub_client_sample_mqtt example.

I believe the issue is that the wifi_config_t structure is a local variable in the initialise_wifi() function call.

We experienced that the example uses the CONFIG_WIFI_SSID and password and this will attach the WIFI correctly if you configured the correct strings in your compile.  If you change the wifi_config_t structure's ssid and password values in that function, and then later perform an attach to WIFI, it does not work, and worse, it hangs waiting for the event.

I was pointed to an example in the esp-idf repository:

https://github.com/espressif/esp-idf/blob/master/examples/provisioning/softap_prov/main/app_prov_handlers.c#L78

The way the WIFI is initialized here, the wifi_config_t structure is a global.  They chose to allocate it, versus just declare it, however I feel either way would work.  I feel this fixes the problem because the WIFI is connected from an entirely different function, and therefore the wifi_config_t structure which was a function local variable, is no longer valid.  My assumption is that the static compilation of this function causes the memory for it to be set aside.

That example also does not have a call to esp_wifi_set_storage().  I do not feel that may be the problem, however I removed it and also made my wifi_config_t structure a global and this fixed it so that I could dynamically update my ssid and password and be able to attach.

Please look into this issue and if you agree that the wifi initialization path should be changed, it will be helpful for anyone in the future who uses that example.

mahavirj, 2019-01-31:
@rtmistler 

For some reason I am still unable to understand this issue or reasoning behind why `wifi_config` needs to be static or global.

- Regarding dynamic configuration of ssid/password, can you please try attached [azure_main.zip](https://github.com/espressif/esp-azure/files/2815808/azure_main.zip) file and see if that is what you are trying to achieve ?
- Regarding `esp_wifi_set_storage()` API, it decides whether WiFi configuration needs to written to NVS partition (flash) or not. You may refer to documentation at, https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv220esp_wifi_set_storage14wifi_storage_t

rtmistler, 2019-01-31:
Hi,

I will try your example and also try to explain the issue more clearly.

I do have to admit that it will be a few days before I can do this.

We are working towards a deadline which has been passed, and therefore are
day to day with getting the application ready for that activity.

Please bear with me. Likely sometime next week.

Best Regards,

-RT

mahavirj, 2019-02-14:
@rtmistler It will be really helpful if you could validate above test application and confirm the fix.

rtmistler, 2019-02-14:
Hi,

The difficulty I've run into is that your version of "dynamic" is not
exactly what we have.

With your sample code, the strings are still defined at compile time.

With our application the strings literally come over a UART attached
protocol.

I'm hard pressed to modify one of your examples where it allows me
something like a console that takes in any string which then would be the
SSID and password.

What I can say is that defining the wifi_config_t local to the function was
not the best option, but making it global scope did work correctly.

Best,

- RT

mahavirj, 2019-02-15:
@rtmistler  Sounds good, I will classify this as use-case driven, and same do not apply to `default` examples in `esp-azure`, so will keep them as they are right now.

closed 2019-02-15 by mahavirj