espressif/esp-azure #6: esp-azure needs to be split into two repositories

seank-com, 2018-08-29:
## Problem:

When Espressif took over the port of the [Azure IoT SDK for Esp32](https://github.com/Azure/azure-iot-pal-esp32) the layout of the port was updated as a sample. Consequently when developers try to add the repo as a component to their projects, it doesn’t build and requires significant changes to get it to work.
 
The standard ESP32 project looks like this
 
components/
esp-idf/
main/
Makefile
 
- Project code goes in main/ 
- You run ```git submodule add https://github.com/espressif/esp-idf.git``` to add the esp-idf as a submodule.
 - Finally for any other libraries you go into the components folder and ```git submodule add``` the components.

For the old azure iot sdk you used to do the following:
 
```bash
$ cd components
$ git submodule add https://github.com/Azure/azure-iot-pal-esp32
 ```

Now that Espressif has taken over, you’d think that doing this instead would just work:
 
```bash
$ cd components
$ git submodule add https://github.com/espressif/esp-azure
```
 
Unfortunately this doesn’t work because the structure of the esp-azure repo is a demo and not a component. Ideally, you’d want a git repo that was just the contents of the azure_iot folder in the components folder of the current repo. Git does not allow you to add a partial repo as a submodule (I looked) so as it is right now, the developer is left with creating their your own repo and copying files from the the [espressif/esp-azure](https://github.com/espressif/esp-azure) repo. This is not a good pattern, because if Espressif or Microsoft should happen to find a security vulnerability and update the repo, the developer cannot just git pull an update but instead will have to copy again by hand.
 
## Proposed change:

- Create a new repo (espressif/azure-iot) that has the contents of the azure_iot folder in the current repo.
- Update the current repo ([espressif/esp-azure](https://github.com/espressif/esp-azure)) to add this new repo as a submodule in the contents folder.

This way the sample continues to work, but developers working on their own projects can add the espressif/azure-iot repo as a submodule.

igrr, 2018-08-29:
Another option might be to move the component part to the top level of this repository and put an example into a subdirectory. This way it is still one repo (hence easier to update) but it can be cloned or added as a submodule into project directory.
One example of such layout is in https://github.com/igrr/libnmea-esp32.

sukeshak, 2018-08-29:
Looks like aws_iot is available as a component under the main tree. 
Is there a reason why Azure support should be done differently?

igrr, 2018-08-29:
Eventually, aws_iot component will also be split out  (as all other non-essential components). This is pending some changes on IDF side to make such external components easier to discover and install.

sukeshak, 2018-08-29:
Sounds good. I am working on a product using ESP32 and Azure thus the question.

ustccw, 2018-08-30:
@seank-com the following structure is not recommended:
```
components/
esp-idf/
main/
Makefile
```
because [esp-azure](https://github.com/espressif/esp-azure) supports all ESP platform, not just esp-idf.

ustccw, 2018-08-30:
split [esp-azure](https://github.com/espressif/esp-azure) to `espressif/azure-iot` and `this example`, this is a question worth discussing. 
one side, it is not easy to maintain, when updating `azure_iot` or some people doubt at `azure-iot` meanings. 
on the other hand, a new project starts with `esp-azure`, not need to copy some code. just  `export IDF_PATH=~/esp/esp-idf` or `export IDF_PATH=~/esp/ESP8266_RTOS_SDK`.  current repo works like this way.

seank-com, 2018-08-31:
@ustccw, thanks I was pointing out an example from my experience working on Esp32 which so far has been entirely Esp-idf, what other Esp platforms are there for the Esp32?

Also, I don't understand your second comment, someone starting a new project can't just use esp-azure because you wouldn't want them to push their project back into the repo would you? Surely, you are not proposing that someone start a project without source control.

sukeshak, 2018-08-31:
Arduino based programming is more popular in makers side of things. I like ESP-IDF but library/driver availability is better with Arduino.

So for now using ESP-IDF with Arduino as a component, which is best of both worlds 😊

ustccw, 2018-09-03:
@seank-com  maybe my expression is not clear, what i mean is that esp-azure can port esp-idf platform and ESP8266_RTOS_SDK platform currently.
if esp-idf as a submodule, then here is no reason for ESP8266_RTOS_SDK not as a submodule. it would make this repo complex and hard to maintain.

markrad, 2018-10-04:
I was able to circumvent this issue by simply using the EXTRA_COMPONENT_DIRS variable in my project Makefile. You just point it at <your path>/esp-azure/components/azure_iot. I've included azure-iot as a Git submodule in my repository.

mahavirj, 2018-12-14:
Repository is now restructured (such that it is easier to add upon more examples), feedback always welcome.