espressif/esp-azure #13: Connection Not Accepted error

ms-iotkithol-jp, 2018-10-16:
Hi,

My device can't send message. It seems that connection error happens but I confirmed that my connection-string is valid. I specified connection string in make manuconfig and it is stored in sdkconfig like.

CONFIG_WIFI_SSID="W04_10B1F8FDB"
CONFIG_WIFI_PASSWORD="_q0---i8_"
CONFIG_IOTHUB_CONNECTION_STRING="HostName=_iothub_.azure-devices.net;DeviceId=_esp32_;SharedAccessKey=_YbI2K----oduPY=_"

-- make monitor log --
Sleeping for 5
-> 14:39:13 CONNECT | VER: 4 | KEEPALIVE: 240 | FLAGS: 192 | USERNAME: egiothub201809.azure-devices.net/esp32/?api-version=2017-11-08-preview&DeviceClientType=iothubclient%2f1.2.8%20(native%3b%20freertos%3b%20esp%20platform) | PWD: XXXX | CLEAN: 0
Error: Time:Tue Oct 16 14:39:18 2018 File:/home/hirosho/develop/esp32/esp-azure/components/azure_iot/azure/iothub_client/src/iothubtransport_mqtt_common.c Func:mqtt_operation_complete_callback Line:1567 Connection Not Accepted: 0x5: Not Authorized
------

What's wrong?
I saw following message so Wi-Fi connection seems to be succesful.

I (381) azure: Setting WiFi configuration SSID W04_10B1F8FDB...
I (461) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (461) wifi: mode : sta (30:ae:a4:02:4a:50)
I (941) wifi: n:4 0, o:1 0, ap:255 255, sta:4 0, prof:1
I (1931) wifi: state: init -> auth (b0)
I (2931) wifi: state: auth -> init (200)
I (2931) wifi: n:4 0, o:4 0, ap:255 255, sta:4 0, prof:1
I (5471) wifi: n:4 0, o:4 0, ap:255 255, sta:4 0, prof:1
I (5471) wifi: state: init -> auth (b0)
I (5471) wifi: state: auth -> assoc (0)
I (5481) wifi: state: assoc -> run (10)
I (5521) wifi: connected with W04_10B1F8FDB, channel 4
I (5521) wifi: pm start, type: 1

I (6281) event: sta ip: 192.168.100.115, mask: 255.255.255.0, gw: 192.168.100.1
I (6281) azure: Connected to AP success!

File:/home/hirosho/develop/esp32/esp-azure/main/iothub_client_sample_mqtt.c Compile Time:Oct 16 2018 15:34:04
Initializing SNTP
ESP platform sntp inited!
Time is not set yet. Connecting to WiFi and getting time over NTP. timeinfo.tm_year:70
Waiting for system time to be set... tm_year:0[times:1]
I (8301) platform: The current date/time in Shanghai is: Tue Oct 16 14:37:36 2018
IoTHubClient_LL_SetMessageCallback...successful.

FHFS, 2018-11-09:
can you try to connect to the Azure IoT Hub after the SNTP has set the time? [simmilar issue](https://github.com/Azure/azure-iot-sdk-c/issues/373)

mahavirj, 2018-12-19:
@ms-iotkithol-jp Request to give a retry, codebase is now updated. If you run into any issues, please let us know.