espressif/esp-azure #15: Provisioning Steps/Doc Requied

mnana3a, 2018-12-19:
Amazing Work, would love to try the provisioning example. but i can't get it to work what files needs to be added , certs. etc.

mahavirj, 2018-12-19:
@mnana3a 
Added instructions with https://github.com/espressif/esp-azure/commit/01e98ac4ff207753fc6fc442fa24854ce1f6e0ab. Essentially device certificate and device private key will be required to drop into `main/certs` directory within `provisioning` example. Let me know if you run into any issues.

mnana3a, 2018-12-19:
@mahavirj 
Thanks for Adding the Docs. but still can't get it to work on my esp8266. i think the only difference here is the `esp_tls` component which isn't available for esp8266. can't it work somehow using the old `openssl` wrappers?

jitin17, 2018-12-20:
@mnana3a We are in the process of adding esp-tls in ESP8266_RTOS_SDK, which might take couple of days to reflect. We'll keep you posted on the same. Thanks.

mnana3a, 2018-12-20:
@jitin17 @mahavirj 
ok now that `esp_tls` is implemented into esp8266_RTOS, i've tried again to use azure dps but i get this error. any help would be much appreciated.
```
E (13198) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (13191) esp-tls: Failed to verify peer certificate!
I (13194) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA
```

jitin17, 2018-12-20:
@mnana3a Do you face the similar issue with esp32?

mnana3a, 2018-12-20:
currently I don't have any esp32 module with me to check

Regards,
Mohamed

jitin17, 2018-12-20:
@mnana3a I suspect there is some issue with your leaf certificate. Did you follow the steps specified  [here](https://github.com/Azure/azure-iot-sdk-c/blob/350b51f5abaedc975dae5419ad1fa4add7635fd2/tools/CACertificates/CACertificateOverview.md).

mnana3a, 2018-12-20:
can you provide me with a cert that works for sure that I can use for testing. thanks

Regards,
Mohamed

mnana3a, 2018-12-20:
also yes that is the steps I followed to produce the leaf cert

jitin17, 2018-12-21:
@mnana3a We encountered a similar issue on esp8266, and it is due to some memory constraints.
As a temporary workaround, you can remove `a/examples/prov_dev_client_ll_sample/main/prov_dev_client_ll_sample.c` from main/component.mk.
Meanwhile, we will try to optimise certain things.

mnana3a, 2018-12-21:
@jitin17  so is there a way to do provision right now using esp8266. is it possible using symmetric keys. Thanks

jitin17, 2018-12-21:
> @jitin17 so is there a way to do provision right now using esp8266. is it possible using symmetric keys. Thanks

You can remove `CFLAGS += -DSET_TRUSTED_CERT_IN_SAMPLES` form main/component.mk as a workaround. This will work but won't verify the Trusted CA Certificate.

mnana3a, 2018-12-21:
Thanks, now it works. hope you can get the CA verification working on esp8266 soon.

closed 2018-12-21 by mnana3a