espressif/esp-azure #16: Compilation fails with latest official ESP-IDF SDK

Patrik-Berglund, 2018-12-21:
Hi,

The build fails for the latest offical ESP-IDF SDK, 3.1.1. https://github.com/espressif/esp-idf/releases/tag/v3.1.1

The compilation passes when using the latest master branch of the ESP-IDF. For our usecase we need a stable ESP-IDF, when do you think that will be available?

```
tlsio_esp_tls.c: In function 'tlsio_esp_tls_open_async':
tlsio_esp_tls.c:317:57: error: 'esp_tls_cfg_t {aka struct esp_tls_cfg}' has no member named 'clientcert_pem_buf'
                             tls_io_instance->esp_tls_cfg.clientcert_pem_buf = (unsigned char *)tls_io_instance->options.x509_cert;
                                                         ^
tlsio_esp_tls.c:318:57: error: 'esp_tls_cfg_t {aka struct esp_tls_cfg}' has no member named 'clientcert_pem_bytes'
                             tls_io_instance->esp_tls_cfg.clientcert_pem_bytes = strlen(tls_io_instance->options.x509_cert) + 1;
                                                         ^
tlsio_esp_tls.c:319:57: error: 'esp_tls_cfg_t {aka struct esp_tls_cfg}' has no member named 'clientkey_pem_buf'
                             tls_io_instance->esp_tls_cfg.clientkey_pem_buf = (unsigned char *)tls_io_instance->options.x509_key;
                                                         ^
tlsio_esp_tls.c:320:57: error: 'esp_tls_cfg_t {aka struct esp_tls_cfg}' has no member named 'clientkey_pem_bytes'
                             tls_io_instance->esp_tls_cfg.clientkey_pem_bytes = strlen(tls_io_instance->options.x509_key) + 1;
                                                         ^
tlsio_esp_tls.c:325:75: error: assignment of read-only member 'cacert_pem_bytes'
                             tls_io_instance->esp_tls_cfg.cacert_pem_bytes = strlen(tls_io_instance->options.trusted_certs) + 1;
                                                                           ^
tlsio_esp_tls.c: In function 'tlsio_esp_tls_dowork':
tlsio_esp_tls.c:470:26: error: implicit declaration of function 'esp_tls_conn_new_async' [-Werror=implicit-function-declaration]
             int result = esp_tls_conn_new_async(tls_io_instance->hostname, strlen(tls_io_instance->hostname), tls_io_instance->port, &tls_io_instance->esp_tls_cfg, tls_io_instance->esp_tls_handle)
```

jitin17, 2018-12-21:
@Patrik-Berglund Latest changes in `esp-tls` component, which is required to run this example, are not a part of release 3.1.1.  It is supported in release 3.2 and on the master branch. It might take a couple of weeks to backport it to release 3.1.
`esp-tls` is a generic TLS layer, as of now, it supports mbedtls and support for wolfssl is in the pipeline.

mahavirj, 2018-12-25:
@Patrik-Berglund 

Yes `release/v3.2` branch is good candidate for testing. Official ESP-IDF v3.2 will be tagged in next few weeks from same branch.

If you run into any issues with long term stability testing, please feel free to notify us.

mahavirj, 2018-12-31:
Closing (reconfirmed build compilation with `release/v3.2` and `master`)

closed 2018-12-31 by mahavirj