espressif/esp-azure #17: Fatal exception 29(StoreProhibitedCause) when connecting to Azure with ESP8266

naakar, 2018-12-23:
Hi

I'll get next exeption when trying to connect Azure IoT hub.
Is that RAM related error or some else ?

Br Kari

Wemos mini d1
Arduino IDE 1.8.8
Azure libraries 1.0.45

Example code from:
git clone https://github.com/Azure-Samples/iot-hub-feather-huzzah-client-app.git

Also if I compile sensor simulation True, compiler gives: segmenatation error ??

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
Serial successfully inited.
Do you need re-input your credential information?(Auto skip this after 5 seconds)[Y/n]
You input nothing, skip...
Attempting to connect to SSID: mokkula_925936
.
mode : sta(60:01:94:64:cd:de) + softAP(62:01:94:64:cd:de)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
Fatal exception 29(StoreProhibitedCause):
epc1=0x4000e1c3, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000018, depc=0x00000000

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3ffffad0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffc70:  3fff172c 00000224 00000224 4010020c  
3ffffc80:  3ffffd40 3fff2d5c 00000008 401006dc

naakar, 2018-12-25:
Hi mnana3a

Done Serial.setDebugOutput(false) 
but still exeption is happening, 
could that happen if connection string primary key is wrong ? 

Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
Serial successfully inited.
Do you need re-input your credential information?(Auto skip this after 5 seconds)[Y/n]
You input nothing, skip...
Attempting to connect to SSID: mokkula_925936
.

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3ffffad0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffc70:  0104a8c0 00000010 00000001 4021d45c  
3ffffc80:  0104a8c0 00ffffff 0104a8c0 feefeffe  
3ffffc90:  00000000 00000000 0000001f 40104bfd  
3ffffca0:  4000050c 00000001 00000010 00000003

naakar, 2019-01-09:
Hi
Has anyone got Wemos mini d1 working with Azure ?
Br Kari

TheGreenToaster, 2019-08-18:
@naakar I was able to get the esp8266 sample from the AzureIoTHub Arduino library running on a Wemo D1 mini.

I was getting the same error until i put in the correct connection string.

sengonik, 2020-06-05:
Fatal exception 29(StoreProhibitedCause):
On my case of Wemos D1 mini, I 've got this messages from serial port due to my program looping to empty void. Just correctly the program void to make sure there is doesn't go through it.