espressif/esp-azure #18: IoT Hub connection not restored when connection to the WIFI AP is lost and then recovered

Patrik-Berglund, 2018-12-26:
ESP-IDF: branch release/v3.2
azure-iot-sdk-c: 2018-12-13

After the ESP32 has connected to the WIFI AP and Azure IoT Hub the WIFI AP is turned off, and then on again after a short while. The connection towards the IoT Hub is not restored after reconnecting to the WIFI AP.

Example code:
https://github.com/Patrik-Berglund/esp32-azure-iot

WIFI AP connection lost at: (43770)
WIFI AP connection recovered at: (70670)

Error Log
```
patrik@ubuntu:~/projects/esp-azure/examples/iothub_client_sample_mqtt$ make  monitor
Toolchain path: /home/patrik/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /home/patrik/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6628
load:0x40078000,len:10976
load:0x40080400,len:7252
entry 0x40080780
I (28) boot: ESP-IDF v3.3-dev-159-ga9eb37e 2nd stage bootloader
I (29) boot: compile time 21:32:00
I (29) boot: Enabling RNG early entropy source...
I (34) qio_mode: Enabling default flash chip QIO
I (40) boot: SPI Speed      : 40MHz
I (44) boot: SPI Mode       : QIO
I (48) boot: SPI Flash Size : 4MB
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x25934 (153908) map
I (120) esp_image: segment 1: paddr=0x0003595c vaddr=0x3ff80000 size=0x00000 (     0) load
I (121) esp_image: segment 2: paddr=0x00035964 vaddr=0x3ff80000 size=0x00000 (     0) load
I (127) esp_image: segment 3: paddr=0x0003596c vaddr=0x3ffb0000 size=0x02e8c ( 11916) load
I (138) esp_image: segment 4: paddr=0x00038800 vaddr=0x3ffb2e8c size=0x00000 (     0) load
I (145) esp_image: segment 5: paddr=0x00038808 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/patrik/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (154) esp_image: segment 6: paddr=0x00038c10 vaddr=0x40080400 size=0x07400 ( 29696) load
I (168) esp_image: segment 7: paddr=0x00040018 vaddr=0x400d0018 size=0x8f38c (586636) map
I (257) esp_image: segment 8: paddr=0x000cf3ac vaddr=0x40087800 size=0x09a64 ( 39524) load
0x40087800: ppTxPkt at ??:?

I (263) esp_image: segment 9: paddr=0x000d8e18 vaddr=0x400c0000 size=0x00000 (     0) load
I (263) esp_image: segment 10: paddr=0x000d8e20 vaddr=0x50000000 size=0x00000 (     0) load
I (272) esp_image: segment 11: paddr=0x000d8e28 vaddr=0x50000000 size=0x00000 (     0) load
I (285) boot: Loaded app from partition at offset 0x10000
I (287) boot: Disabling RNG early entropy source...
I (293) cpu_start: Pro cpu up.
I (297) cpu_start: Starting app cpu, entry point is 0x40081028
0x40081028: call_start_cpu1 at /home/patrik/esp/esp-idf/components/esp32/cpu_start.c:245

[... 3940 lines cut ...]
E (82170) esp-tls: read error :-76:
E (82180) esp-tls: read error :-76:
E (82190) esp-tls: read error :-76:
E (82200) esp-tls: read error :-76:
E (82210) esp-tls: read error :-76:
E (82220) esp-tls: read error :-76:
E (82230) esp-tls: read error :-76:
E (82240) esp-tls: read error :-76:
E (82250) esp-tls: read error :-76:
E (82260) esp-tls: read error :-76:
E (82270) esp-tls: read error :-76:
E (82280) esp-tls: read error :-76:
E (82290) esp-tls: read error :-76:
E (82300) esp-tls: read error :-76:
E (82310) esp-tls: read error :-76:
E (82320) esp-tls: read error :-76:
E (82330) esp-tls: read error :-76:
E (82340) esp-tls: read error :-76:
E (82350) esp-tls: read error :-76:
E (82360) esp-tls: read error :-76:
E (82370) esp-tls: read error :-76:
E (82380) esp-tls: read error :-76:

patrik@ubuntu:~/projects/esp-azure/examples/iothub_client_sample_mqtt$ 
```

jitin17, 2018-12-31:
@Patrik-Berglund In case of AP connection loss, there is a need to re-connect to Azure IoT Hub.
And when there is an AP connection loss, we observe `errno` to be `EHOSTUNREACH`. So, we have added enhancement in `iothub_client_sample_mqtt` example to re-connect to Azure IoT Portal upon observing `errno` as `EHOSTUNREACH`. And we wait for WiFi STA state to be connected before connecting to Azure IoT Portal. Please refer to changes in `iothub_client_sample_mqtt` example to add corresponding changes in your example.  Let us know whether this works or not.

Patrik-Berglund, 2019-01-01:
@jitin17 

The IoT Hub SDK will it self handle the re-connection. I have made a small update to the "esp-azure" to make it work. 

I will prepare a pull-request with the changes needed.

jitin17, 2019-02-15:
@Patrik-Berglund We have seen that [azure-iot-sdk-c](https://github.com/Azure/azure-iot-sdk-c) handles internet re-connection and connection to WiFi AP loss with the help of retry control handle. But there is a bug in `azure-iot-sdk-c` which causes crash; we have found the solution to it and will raise a PR soon. Thanks.

Patrik-Berglund, 2019-02-15:
Great! 

We switched to using the mbedtls and berkeley sockets based PAL provided by the IoT Hub C SDK instead. Required only minimal changes to their code.

closed 2019-03-01 by mahavirj