espressif/esp-azure #19: cmake support?

dtboy1995, 2018-12-27:
Do you plan to support cmake builds? Thank you

dtboy1995, 2018-12-27:
clion log
```
"C:\Program Files\JetBrains\CLion 2018.3.2\bin\cmake\win\bin\cmake.exe" -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - NMake Makefiles" D:\esp\hello_world
CMake Error at D:/msys32/home/dtboy/esp/esp-idf/components/azure-iot/CMakeLists.txt:6 (project):
  project command is not scriptable
Call Stack (most recent call first):
  D:/msys32/home/dtboy/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:151 (include)
  D:/msys32/home/dtboy/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:159 (expand_component_requirements)
  D:/msys32/home/dtboy/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:183 (expand_component_requirements)

-- Component names: main app_trace xtensa-debug-module app_update spi_flash bootloader_support mbedtls lwip vfs ethernet tcpip_adapter micro-ecc bootloader bt nvs_flash coap console cxx driver esp-tls nghttp esp32 log pthread smartconfig_ack wpa_supplicant esp_adc_cal esp_http_client esp_https_ota esptool_py expat fatfs wear_levelling sdmmc freertos heap idf_test jsmn json libsodium mdns newlib openssl partition_table soc spiffs ulp
-- Component paths: D:/esp/hello_world/main;D:/msys32/home/dtboy/esp/esp-idf/components/app_trace;D:/msys32/home/dtboy/esp/esp-idf/components/xtensa-debug-module;D:/msys32/home/dtboy/esp/esp-idf/components/app_update;D:/msys32/home/dtboy/esp/esp-idf/components/spi_flash;D:/msys32/home/dtboy/esp/esp-idf/components/bootloader_support;D:/msys32/home/dtboy/esp/esp-idf/components/mbedtls;D:/msys32/home/dtboy/esp/esp-idf/components/lwip;D:/msys32/home/dtboy/esp/esp-idf/components/vfs;D:/msys32/home/dtboy/esp/esp-idf/components/ethernet;D:/msys32/home/dtboy/esp/esp-idf/components/tcpip_adapter;D:/msys32/home/dtboy/esp/esp-idf/components/micro-ecc;D:/msys32/home/dtboy/esp/esp-idf/components/bootloader;D:/msys32/home/dtboy/esp/esp-idf/components/bt;D:/msys32/home/dtboy/esp/esp-idf/components/nvs_flash;D:/msys32/home/dtboy/esp/esp-idf/components/coap;D:/msys32/home/dtboy/esp/esp-idf/components/console;D:/msys32/home/dtboy/esp/esp-idf/components/cxx;D:/msys32/home/dtboy/esp/esp-idf/components/driver;D:/msys32/home/dtboy/esp/esp-idf/components/esp-tls;D:/msys32/home/dtboy/esp/esp-idf/components/nghttp;D:/msys32/home/dtboy/esp/esp-idf/components/esp32;D:/msys32/home/dtboy/esp/esp-idf/components/log;D:/msys32/home/dtboy/esp/esp-idf/components/pthread;D:/msys32/home/dtboy/esp/esp-idf/components/smartconfig_ack;D:/msys32/home/dtboy/esp/esp-idf/components/wpa_supplicant;D:/msys32/home/dtboy/esp/esp-idf/components/esp_adc_cal;D:/msys32/home/dtboy/esp/esp-idf/components/esp_http_client;D:/msys32/home/dtboy/esp/esp-idf/components/esp_https_ota;D:/msys32/home/dtboy/esp/esp-idf/components/esptool_py;D:/msys32/home/dtboy/esp/esp-idf/components/expat;D:/msys32/home/dtboy/esp/esp-idf/components/fatfs;D:/msys32/home/dtboy/esp/esp-idf/components/wear_levelling;D:/msys32/home/dtboy/esp/esp-idf/components/sdmmc;D:/msys32/home/dtboy/esp/esp-idf/components/freertos;D:/msys32/home/dtboy/esp/esp-idf/components/heap;D:/msys32/home/dtboy/esp/esp-idf/components/idf_test;D:/msys32/home/dtboy/esp/esp-idf/components/jsmn;D:/msys32/home/dtboy/esp/esp-idf/components/json;D:/msys32/home/dtboy/esp/esp-idf/components/libsodium;D:/msys32/home/dtboy/esp/esp-idf/components/mdns;D:/msys32/home/dtboy/esp/esp-idf/components/newlib;D:/msys32/home/dtboy/esp/esp-idf/components/openssl;D:/msys32/home/dtboy/esp/esp-idf/components/partition_table;D:/msys32/home/dtboy/esp/esp-idf/components/soc;D:/msys32/home/dtboy/esp/esp-idf/components/spiffs;D:/msys32/home/dtboy/esp/esp-idf/components/ulp
-- Adding linker script D:/esp/hello_world/cmake-build-debug/esp32/esp32_out.ld
-- Adding linker script D:/msys32/home/dtboy/esp/esp-idf/components/esp32/ld/esp32.common.ld
-- Adding linker script D:/msys32/home/dtboy/esp/esp-idf/components/esp32/ld/esp32.rom.ld
-- Adding linker script D:/msys32/home/dtboy/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script D:/msys32/home/dtboy/esp/esp-idf/components/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script D:/msys32/home/dtboy/esp/esp-idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld
-- Component libraries: ;main;app_trace;xtensa-debug-module;app_update;spi_flash;bootloader_support;mbedtls;lwip;vfs;ethernet;tcpip_adapter;micro-ecc;bt;nvs_flash;coap;console;cxx;driver;esp-tls;nghttp;esp32;log;pthread;smartconfig_ack;wpa_supplicant;esp_adc_cal;esp_http_client;esp_https_ota;expat;fatfs;wear_levelling;sdmmc;freertos;heap;idf_test;jsmn;json;libsodium;mdns;newlib;openssl;soc;spiffs;ulp
-- Configuring done
-- Generating done
-- Build files have been written to: D:/esp/hello_world/cmake-build-debug

```

dtboy1995, 2018-12-27:
@mahavirj , deeply grateful, let me try

dtboy1995, 2018-12-28:
@mahavirj 
```
"C:\Program Files\JetBrains\CLion 2018.3.2\bin\cmake\win\bin\cmake.exe" -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - NMake Makefiles" D:\msys32\home\dtboy\esp\esp-idf\components\esp-azure\examples\iothub_client_sample_mqtt
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.13.0.windows.1") 
CMake Warning at D:/msys32/home/dtboy/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:118 (message):
  Component D:/msys32/home/dtboy/esp/esp-idf/components/esp-azure contains
  old-style component.mk but no CMakeLists.txt.  Component will be skipped.
Call Stack (most recent call first):
  D:/msys32/home/dtboy/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:167 (components_find_all)

-- Component names: port mbedtls lwip vfs ethernet tcpip_adapter esp-tls nghttp main app_trace xtensa-debug-module app_update spi_flash bootloader_support micro-ecc bootloader bt nvs_flash coap console cxx driver esp32 log pthread smartconfig_ack wpa_supplicant esp_adc_cal esp_http_client esp_https_ota esptool_py expat fatfs wear_levelling sdmmc freertos heap idf_test jsmn json libsodium mdns newlib openssl partition_table soc spiffs ulp
-- Component paths: D:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/examples/iothub_client_sample_mqtt/cmake-build-debug/../../../port;D:/msys32/home/dtboy/esp/esp-idf/components/mbedtls;D:/msys32/home/dtboy/esp/esp-idf/components/lwip;D:/msys32/home/dtboy/esp/esp-idf/components/vfs;D:/msys32/home/dtboy/esp/esp-idf/components/ethernet;D:/msys32/home/dtboy/esp/esp-idf/components/tcpip_adapter;D:/msys32/home/dtboy/esp/esp-idf/components/esp-tls;D:/msys32/home/dtboy/esp/esp-idf/components/nghttp;D:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/examples/iothub_client_sample_mqtt/main;D:/msys32/home/dtboy/esp/esp-idf/components/app_trace;D:/msys32/home/dtboy/esp/esp-idf/components/xtensa-debug-module;D:/msys32/home/dtboy/esp/esp-idf/components/app_update;D:/msys32/home/dtboy/esp/esp-idf/components/spi_flash;D:/msys32/home/dtboy/esp/esp-idf/components/bootloader_support;D:/msys32/home/dtboy/esp/esp-idf/components/micro-ecc;D:/msys32/home/dtboy/esp/esp-idf/components/bootloader;D:/msys32/home/dtboy/esp/esp-idf/components/bt;D:/msys32/home/dtboy/esp/esp-idf/components/nvs_flash;D:/msys32/home/dtboy/esp/esp-idf/components/coap;D:/msys32/home/dtboy/esp/esp-idf/components/console;D:/msys32/home/dtboy/esp/esp-idf/components/cxx;D:/msys32/home/dtboy/esp/esp-idf/components/driver;D:/msys32/home/dtboy/esp/esp-idf/components/esp32;D:/msys32/home/dtboy/esp/esp-idf/components/log;D:/msys32/home/dtboy/esp/esp-idf/components/pthread;D:/msys32/home/dtboy/esp/esp-idf/components/smartconfig_ack;D:/msys32/home/dtboy/esp/esp-idf/components/wpa_supplicant;D:/msys32/home/dtboy/esp/esp-idf/components/esp_adc_cal;D:/msys32/home/dtboy/esp/esp-idf/components/esp_http_client;D:/msys32/home/dtboy/esp/esp-idf/components/esp_https_ota;D:/msys32/home/dtboy/esp/esp-idf/components/esptool_py;D:/msys32/home/dtboy/esp/esp-idf/components/expat;D:/msys32/home/dtboy/esp/esp-idf/components/fatfs;D:/msys32/home/dtboy/esp/esp-idf/components/wear_levelling;D:/msys32/home/dtboy/esp/esp-idf/components/sdmmc;D:/msys32/home/dtboy/esp/esp-idf/components/freertos;D:/msys32/home/dtboy/esp/esp-idf/components/heap;D:/msys32/home/dtboy/esp/esp-idf/components/idf_test;D:/msys32/home/dtboy/esp/esp-idf/components/jsmn;D:/msys32/home/dtboy/esp/esp-idf/components/json;D:/msys32/home/dtboy/esp/esp-idf/components/libsodium;D:/msys32/home/dtboy/esp/esp-idf/components/mdns;D:/msys32/home/dtboy/esp/esp-idf/components/newlib;D:/msys32/home/dtboy/esp/esp-idf/components/openssl;D:/msys32/home/dtboy/esp/esp-idf/components/partition_table;D:/msys32/home/dtboy/esp/esp-idf/components/soc;D:/msys32/home/dtboy/esp/esp-idf/components/spiffs;D:/msys32/home/dtboy/esp/esp-idf/components/ulp
Creating config file D:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/examples/iothub_client_sample_mqtt/sdkconfig...
-- The ASM compiler identification is GNU
-- Found assembler: D:/msys32/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- Check for working C compiler: D:/msys32/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/msys32/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: D:/msys32/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: D:/msys32/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Adding linker script D:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/examples/iothub_client_sample_mqtt/cmake-build-debug/esp32/esp32_out.ld
-- Adding linker script D:/msys32/home/dtboy/esp/esp-idf/components/esp32/ld/esp32.common.ld
-- Adding linker script D:/msys32/home/dtboy/esp/esp-idf/components/esp32/ld/esp32.rom.ld
-- Adding linker script D:/msys32/home/dtboy/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script D:/msys32/home/dtboy/esp/esp-idf/components/esp32/ld/esp32.rom.libgcc.ld
[... cut ...]
-- Component libraries: ;port;mbedtls;lwip;vfs;ethernet;tcpip_adapter;esp-tls;nghttp;main;app_trace;xtensa-debug-module;app_update;spi_flash;bootloader_support;micro-ecc;bt;nvs_flash;coap;console;cxx;driver;esp32;log;pthread;smartconfig_ack;wpa_supplicant;esp_adc_cal;esp_http_client;esp_https_ota;expat;fatfs;wear_levelling;sdmmc;freertos;heap;idf_test;jsmn;json;libsodium;mdns;newlib;openssl;soc;spiffs;ulp
-- Configuring done
-- Generating done
-- Build files have been written to: D:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/examples/iothub_client_sample_mqtt/cmake-build-debug

[Finished]
```

dtboy1995, 2018-12-28:
```

D:\msys32\home\dtboy\esp\esp-idf\components\esp-azure\examples\iothub_client_sample_mqtt>idf.py build
Running ninja in directory D:\msys32\home\dtboy\esp\esp-idf\components\esp-azure\examples\iothub_client_sample_mqtt\build
Executing "ninja all"...
[59/542] Building C object port/CMakeFiles/port.dir/src/agenttime_esp.c.obj
FAILED: port/CMakeFiles/port.dir/src/agenttime_esp.c.obj
D:\msys32\opt\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DHAVE_CONFIG_H -DHSM_TYPE_SAS_TOKEN -DHSM_TYPE_X509 -DIDF_V
ER=\"v3.1.1\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUSE_LWIP_SOCKET_FOR_AZURE_IOT -Iconfig -ID:/msys32/home/dtboy/esp/esp-id
f/components/esp-azure/port/inc -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/certs -ID:/msys32/home/dtboy/es
p/esp-idf/components/esp-azure/azure-iot-sdk-c/c-utility/inc -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/c-
utility/inc/azure_c_shared_utility -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/c-utility/pal/inc -ID:/msys3
2/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/c-utility/pal/freertos -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-
azure/azure-iot-sdk-c/c-utility/pal/generic -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/iothub_client/inc -
ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/serializer/inc -ID:/msys32/home/dtboy/esp/esp-idf/components/esp
-azure/azure-iot-sdk-c/umqtt/inc -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/umqtt/inc/azure_umqtt_c -ID:/m
sys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/deps/parson -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/a
zure-iot-sdk-c/provisioning_client/inc -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/provisioning_client/adap
ters -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/provisioning_client/deps/utpm/inc -ID:/msys32/home/dtboy/e
sp/esp-idf/components/mbedtls/port/include -ID:/msys32/home/dtboy/esp/esp-idf/components/mbedtls/mbedtls/include -ID:/msys32/home/dtboy
/esp/esp-idf/components/lwip/include/lwip -ID:/msys32/home/dtboy/esp/esp-idf/components/lwip/include/lwip/port -ID:/msys32/home/dtboy/e
sp/esp-idf/components/lwip/include/lwip/posix -ID:/msys32/home/dtboy/esp/esp-idf/components/lwip/apps/ping -ID:/msys32/home/dtboy/esp/e
sp-idf/components/vfs/include -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-tls -ID:/msys32/home/dtboy/esp/esp-idf/components/esp32
/include -ID:/msys32/home/dtboy/esp/esp-idf/components/driver/include -ID:/msys32/home/dtboy/esp/esp-idf/components/tcpip_adapter/inclu
de -ID:/msys32/home/dtboy/esp/esp-idf/components/newlib/platform_include -ID:/msys32/home/dtboy/esp/esp-idf/components/newlib/include -
ID:/msys32/home/dtboy/esp/esp-idf/components/freertos/include -ID:/msys32/home/dtboy/esp/esp-idf/components/app_trace/include -ID:/msys
32/home/dtboy/esp/esp-idf/components/heap/include -ID:/msys32/home/dtboy/esp/esp-idf/components/log/include -ID:/msys32/home/dtboy/esp/
esp-idf/components/soc/esp32/include -ID:/msys32/home/dtboy/esp/esp-idf/components/soc/include -Og -std=gnu99 -ffunction-sections -fdat
a-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-var
iable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-decl
aration -ggdb -Wno-unused-function -Wno-missing-braces -Wno-missing-field-initializers -MD -MT port/CMakeFiles/port.dir/src/agenttime_e
sp.c.obj -MF port\CMakeFiles\port.dir\src\agenttime_esp.c.obj.d -o port/CMakeFiles/port.dir/src/agenttime_esp.c.obj   -c D:/msys32/home
/dtboy/esp/esp-idf/components/esp-azure/port/src/agenttime_esp.c
D:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/port/src/agenttime_esp.c:19:28: fatal error: lwip/apps/sntp.h: No such file or di
rectory
compilation terminated.
[60/542] Building C object port/CMakeFiles/port.dir/src/platform_esp.c.obj
FAILED: port/CMakeFiles/port.dir/src/platform_esp.c.obj
D:\msys32\opt\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DHAVE_CONFIG_H -DHSM_TYPE_SAS_TOKEN -DHSM_TYPE_X509 -DIDF_V
ER=\"v3.1.1\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUSE_LWIP_SOCKET_FOR_AZURE_IOT -Iconfig -ID:/msys32/home/dtboy/esp/esp-id
f/components/esp-azure/port/inc -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/certs -ID:/msys32/home/dtboy/es
p/esp-idf/components/esp-azure/azure-iot-sdk-c/c-utility/inc -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/c-
utility/inc/azure_c_shared_utility -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/c-utility/pal/inc -ID:/msys3
2/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/c-utility/pal/freertos -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-
azure/azure-iot-sdk-c/c-utility/pal/generic -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/iothub_client/inc -
ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/serializer/inc -ID:/msys32/home/dtboy/esp/esp-idf/components/esp
-azure/azure-iot-sdk-c/umqtt/inc -ID:/msys32/home/dtboy/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/umqtt/inc/azure_umqtt_c -ID:/m
[... cut ...]
_conn_new_async' [-Werror=implicit-function-declaration]
             int result = esp_tls_conn_new_async(tls_io_instance->hostname, strlen(tls_io_instance->hostname), tls_io_instance->port, &
tls_io_instance->esp_tls_cfg, tls_io_instance->esp_tls_handle);
                          ^
cc1.exe: some warnings being treated as errors
[64/542] Building C object port/CMakeFiles/port.dir/D_/msys32/home...onents/esp-azure/azure-iot-sdk-c/c-utility/pal/tlsio_options.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

mahavirj, 2018-12-28:
@dtboy1995 Could you pleas help to confirm if your `ESP-IDF` is on either `master` or `release/v3.2` branch?

mahavirj, 2018-12-28:
@dtboy1995 Yes, please use `master` or `release/v3.2` branch of IDF. Official IDFv3.2 is expected in next few weeks.