espressif/esp-azure #20: Setting a connceted device to disabled in IoT Hub causes esp-tls failure

Patrik-Berglund, 2018-12-27:
ESP-IDF: branch release/v3.2
azure-iot-sdk-c: 2018-12-13

Setting a connceted device to disabled in IoT Hub causes esp-tls failure.

Example code:
https://github.com/Patrik-Berglund/esp32-azure-iot

Device disabled in IoT Hub at (247130)

Error Log
````
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6272
ho 0 tail 12 room 4
load:0x40078000,len:10168
ho 0 tail 12 room 4
load:0x40080400,len:6612
entry 0x40080764
I (33) boot: ESP-IDF v3.3-dev-159-ga9eb37e-dirty 2nd stage bootloader
I (33) boot: compile time 18:10:34
I (34) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : DIO
I (47) boot: SPI Flash Size : 4MB
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x25b10 (154384) map
I (128) esp_image: segment 1: paddr=0x00035b38 vaddr=0x3ff80000 size=0x00000 (     0) load
I (128) esp_image: segment 2: paddr=0x00035b40 vaddr=0x3ff80000 size=0x00000 (     0) load
I (135) esp_image: segment 3: paddr=0x00035b48 vaddr=0x3ffb0000 size=0x02e94 ( 11924) load
I (146) esp_image: segment 4: paddr=0x000389e4 vaddr=0x3ffb2e94 size=0x00000 (     0) load
I (153) esp_image: segment 5: paddr=0x000389ec vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/patrikb/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (162) esp_image: segment 6: paddr=0x00038df4 vaddr=0x40080400 size=0x0721c ( 29212) load
I (177) esp_image: segment 7: paddr=0x00040018 vaddr=0x400d0018 size=0x8f304 (586500) map
0x400d0018: _stext at ??:?

I (295) esp_image: segment 8: paddr=0x000cf324 vaddr=0x4008761c size=0x09c48 ( 40008) load
0x4008761c: ppTask at ??:?

I (304) esp_image: segment 9: paddr=0x000d8f74 vaddr=0x400c0000 size=0x00000 (     0) load
I (304) esp_image: segment 10: paddr=0x000d8f7c vaddr=0x50000000 size=0x00000 (     0) load
I (311) esp_image: segment 11: paddr=0x000d8f84 vaddr=0x50000000 size=0x00000 (     0) load
I (324) boot: Loaded app from partition at offset 0x10000
I (326) boot: Disabling RNG early entropy source...
I (332) cpu_start: Pro cpu up.
I (336) cpu_start: Starting app cpu, entry point is 0x40081028
0x40081028: call_start_cpu1 at /home/patrikb/esp/esp-idf/components/esp32/cpu_start.c:245

I (0) cpu_start: App cpu up.
I (346) heap_init: Initializing. RAM available for dynamic allocation:
I (353) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (359) heap_init: At 3FFB8F28 len 000270D8 (156 KiB): DRAM
I (365) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (371) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (378) heap_init: At 40091264 len 0000ED9C (59 KiB): IRAM
I (384) cpu_start: Pro cpu start user code
I (29) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (130) MAIN: Reset reason Core0: 16
I (130) MAIN: Reset reason Core1: 14
I (130) wifi: wifi driver task: 3ffc0abc, prio:23, stack:3584, core=0
I (130) wifi: wifi firmware version: 72057e9
I (130) wifi: config NVS flash: enabled
I (140) wifi: config nano formating: disabled
I (140) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (150) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (170) wifi: Init dynamic tx buffer num: 32
I (180) wifi: Init data frame dynamic rx buffer num: 32
I (180) wifi: Init management frame dynamic rx buffer num: 32
I (180) wifi: Init static rx buffer size: 1600
I (180) wifi: Init static rx buffer num: 10
I (190) wifi: Init dynamic rx buffer num: 32
I (190) MAIN: Setting WiFi configuration SSID CGrind_01...
I (260) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (270) wifi: mode : sta (24:0a:c4:82:ab:48)
I (270) MAIN: SYSTEM_EVENT_STA_START.
W (270) wifi: Haven't to connect to a suitable AP now!
I (280) MAIN: Time UTC:             1970-01-01 00:00:00
I (280) MAIN: WIFI RSSI:            0
I (280) MAIN: Total free bytes:     231120
I (300) MAIN: Total allocated bytes:63512
I (300) MAIN: Minimum free bytes:   225864
I (300) MAIN: Largest free block :  113804
I (300) MAIN: Allocated blocks:     130
I (320) MAIN: Free blocks:          6
I (320) MAIN: Total blocks:         136
I (400) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
I (1050) wifi: state: init -> auth (b0)
I (1060) wifi: state: auth -> assoc (0)
I (1060) wifi: state: assoc -> run (10)
I (1080) wifi: connected with CGrind_01, channel 11
I (1080) wifi: pm start, type: 1

I (1080) MAIN: SYSTEM_EVENT_STA_CONNECTED.
I (5130) event: sta ip: 192.168.1.113, mask: 255.255.255.0, gw: 192.168.1.1
I (5130) MAIN: SYSTEM_EVENT_STA_GOT_IP.
I (5130) MAIN: Connected to AP success!
Info: Initializing SNTP
Info: SNTP initialization complete
-> 17:11:53 CONNECT | VER: 4 | KEEPALIVE: 240 | FLAGS: 192 | USERNAME: cgrind-dev-iothub01.azure-devices.net/240ac482ab48/?api-version=2017-11-08-preview&DeviceClientType=iothubclient%2f1.2.12%20(openssl_compact) | PWD: XXXX | CLEAN: 0
<- 17:11:53 CONNACK | SESSION_PRESENT: true | RETURN_CODE: 0x0
I (7340) IOTHUB: Connection Status Callback: status=0 reason=6
I (7340) IOTHUB: IOTHUB_CLIENT_CONNECTION_OK
I (60330) MAIN: Time UTC:             2018-12-27 17:12:46
I (60330) MAIN: WIFI RSSI:            -56
I (60330) MAIN: Total free bytes:     168420
I (60330) MAIN: Total allocated bytes:125664
I (60350) MAIN: Minimum free bytes:   153652
I (60350) MAIN: Largest free block :  113804
I (60350) MAIN: Allocated blocks:     256
I (60350) MAIN: Free blocks:          17
I (60350) MAIN: Total blocks:         273
I (120380) MAIN: Time UTC:             2018-12-27 17:13:46
I (120380) MAIN: WIFI RSSI:            -57
I (120380) MAIN: Total free bytes:     168420
I (120380) MAIN: Total allocated bytes:125664
I (120400) MAIN: Minimum free bytes:   153652
I (120400) MAIN: Largest free block :  113804
I (120400) MAIN: Allocated blocks:     256
I (120400) MAIN: Free blocks:          17
I (120410) MAIN: Total blocks:         273
I (180430) MAIN: Time UTC:             2018-12-27 17:14:46
I (180430) MAIN: WIFI RSSI:            -57
I (180430) MAIN: Total free bytes:     168632
I (180430) MAIN: Total allocated bytes:125452
I (180450) MAIN: Minimum free bytes:   153652
I (180450) MAIN: Largest free block :  113804
I (180450) MAIN: Allocated blocks:     254
I (180450) MAIN: Free blocks:          19
I (180460) MAIN: Total blocks:         273
I (240490) MAIN: Time UTC:             2018-12-27 17:15:46
I (240490) MAIN: WIFI RSSI:            -56
I (240490) MAIN: Total free bytes:     168632
I (240490) MAIN: Total allocated bytes:125452
I (240510) MAIN: Minimum free bytes:   153652
I (240510) MAIN: Largest free block :  113804
I (240510) MAIN: Allocated blocks:     254
I (240510) MAIN: Free blocks:          19
I (240520) MAIN: Total blocks:         273
E (247130) esp-tls: write error :-80:
Info: Error from SSL_write: -80
-> 17:15:53 PINGREQ
E (247140) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247150) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247160) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247170) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247180) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247190) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247200) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247210) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247220) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247230) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247240) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247250) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247260) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247270) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247280) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247290) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247300) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247310) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (247320) esp-tls: write error :-80:
`´´´

jitin17, 2018-12-31:
@Patrik-Berglund Actually, this is not an issue with esp-tls layer. This condition should be handled in the application itself. When you disable the device in IoT Hub, the `errno` observed is `EKEYREVOKED`. So, you can choose to stop the communication upon observing `errno` as `EKEYREVOKED`.

Patrik-Berglund, 2019-01-01:
@jitin17 

The IoT Hub SDK handles that part for us, it will go in to a retry loop until the device is enabled again.

With a small update to the "esp-azure" it works, I will prepare a pull-request with the changes needed.

Patrik-Berglund, 2019-01-20:
Currently re-running tests with the master branch of azure-iot-sdk-c, there has been some bug fixes since the 2018-12-13 release that might have caused some of the crashes we have seen.

Still using (mbedtls - berkeley version) of the PAL on 10 devices, will run it for a week. If that passes we will run an additional week with poor/intermittent loss of WIFI signal to make sure that the azure-iot-sdk-c always recovers the connection without causing crashes.

jitin17, 2019-03-01:
@Patrik-Berglund We have updated the `azure-iot-sdk-c`, and also fixed `mqtt_client_dowork` crash with PR https://github.com/Azure/azure-iot-sdk-c/pull/875. So, I think this problem should go away as well. Let us know, whether you still face this issue, otherwise we can close it.

Patrik-Berglund, 2019-03-01:
Hi, 
@jitin17 After more testing we went with using the PAL provided by the IoT Hub SDK 
(mbedtls - berkeley version) in our product. 

So we wont do any tests from our side at the moment with the espressif PAL. So this needs to be tested by some one else.

jitin17, 2019-03-01:
@Patrik-Berglund I have tested it, and with the latest HEAD crash is gone. So, I think we can close this issue.

closed 2019-03-04 by mahavirj