espressif/esp-azure #23: Upload to blob storage is not working

jubueche, 2019-02-02:
I am trying to run the example from the iot hub sdk: iothub_client -> samples -> iothub_client_sample_upload_to_bob_mb
I had to change the component.mk file so that it compiles:

# Component Makefile
# Manually inserted: httpi_compact.o
# azure-iot-sdk-c/iothub_client/inc/internal \
# azure-iot-sdk-c/iothub_client/src/blob.o \
# azure-iot-sdk-c/iothub_client/src/iothubtransporthttp.o \

You just need to add these lines to your component.mk file and it should compile.

I haven't changed anything in the example except for the option of using certificates. I am using the standard MSFT certificate (Baltimore). This even fixed another issue. This issue was after a failed connection I would get 403 errors.

Can somebody try to reproduce this?

This is my trace:

0x400d6921: _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1529

0x40160286: _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:72

0x401602c2: vsnprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:41

0x400f4e06: STRING_construct_sprintf at /home/julian/eclipse-workspace/chaze-esp32/components/esp-azure/azure-iot-sdk-c/c-utility/src/strings.c:786

0x400ed89c: IoTHubClient_LL_UploadMultipleBlocksToBlob_Impl at /home/julian/eclipse-workspace/chaze-esp32/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothub_client_ll_uploadtoblob.c:910

0x400ec969: IoTHubClientCore_LL_UploadMultipleBlocksToBlobEx at /home/julian/eclipse-workspace/chaze-esp32/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothub_client_core_ll.c:2595

0x400eb352: IoTHubDeviceClient_LL_UploadMultipleBlocksToBlob at /home/julian/eclipse-workspace/chaze-esp32/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothub_device_client_ll.c:127

0x400914be: upload_to_blob_block at /home/julian/eclipse-workspace/chaze-esp32/components/upload_to_blob_block/upload_to_blob_block.c:113

0x400d4052: azure_task at /home/julian/eclipse-workspace/chaze-esp32/main/main.c:89

0x4008e73d: vPortTaskWrapper at /home/julian/Documents/esp/esp-idf/components/freertos/port.c:403

This is where it happens:
In the upload callback, it sets the data pointer to NULL and the size to 0. After that I get the exception above.

0x400d6921: _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1529

0x40160286: _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:72

0x401602c2: vsnprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:41

0x400f4e06: STRING_construct_sprintf at /home/julian/eclipse-workspace/chaze-esp32/components/esp-azure/azure-iot-sdk-c/c-utility/src/strings.c:786

0x400ed89c: IoTHubClient_LL_UploadMultipleBlocksToBlob_Impl at /home/julian/eclipse-workspace/chaze-esp32/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothub_client_ll_uploadtoblob.c:910

0x400ec969: IoTHubClientCore_LL_UploadMultipleBlocksToBlobEx at /home/julian/eclipse-workspace/chaze-esp32/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothub_client_core_ll.c:2595

0x400eb352: IoTHubDeviceClient_LL_UploadMultipleBlocksToBlob at /home/julian/eclipse-workspace/chaze-esp32/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothub_device_client_ll.c:127

0x400914be: upload_to_blob_block at /home/julian/eclipse-workspace/chaze-esp32/components/upload_to_blob_block/upload_to_blob_block.c:113

0x400d4052: azure_task at /home/julian/eclipse-workspace/chaze-esp32/main/main.c:89

0x4008e73d: vPortTaskWrapper at /home/julian/Documents/esp/esp-idf/components/freertos/port.c:403

jubueche, 2019-02-04:
On the Esp32.

jubueche, 2019-02-04:
In the main.cpp I create a task, yes. I already tried increasing it from (*1024 to 12*1024. Seems like a lot, but I can try increase it more.

jubueche, 2019-02-04:
Quick update: 
This line is where it fails: In strings.c
`length = vsnprintf(buf, maxBufSize, format, arg_list);`

This is what the input to the function looks like prior to failing:
`Args passed to vsnprintf: buf: 0x3ffcade0, maxBufSize: 512, format: {"isSuccess":%s, "statusCode":%d, "statusDescription":"%s"}`

And this is the input to the args list in `arg_list`:

 `((httpResponse < 300) ? "true" : "false"), httpResponse, tmp`

with

`unsigned char * tmp = BUFFER_u_char(responseToIoTHub);` and 
`BUFFER_HANDLE responseToIoTHub = BUFFER_new();`

This is Buffer_new():

```
/* Codes_SRS_BUFFER_07_001: [BUFFER_new shall allocate a BUFFER_HANDLE that will contain a NULL unsigned char*.] */
BUFFER_HANDLE BUFFER_new(void)
{
    BUFFER* temp = (BUFFER*)malloc(sizeof(BUFFER));
    /* Codes_SRS_BUFFER_07_002: [BUFFER_new shall return NULL on any error that occurs.] */
    if (temp != NULL)
    {
        temp->buffer = NULL;
        temp->size = 0;
    }
    return (BUFFER_HANDLE)temp;
}
```

jubueche, 2019-02-04:
Apparently `unsigned char * tmp = BUFFER_u_char(responseToIoTHub);` returns a 0x0 pointer. This makes sense. In Buffer_new() we check if either the handle is NULL or the data size. Since this occurs after the last block was uploaded, the data size is 0 (set in the callback) and we get a NULL pointer. I have included a check for a NULL pointer, but I am not sure what to send as the IoTHubResponse in this case.

jubueche, 2019-02-04:
Ok that fixed the issue.

closed 2019-02-04 by jubueche

jitin17, 2019-02-04:
@jubueche What was the fix?

jubueche, 2019-02-04:
https://github.com/Azure/azure-iot-sdk-c/issues/814 at the bottom.

liudr, 2019-11-25:
I'm trying to compile but failed and added the three lines jubueche mentioned. I keep getting "missing separator" error. Tried iterations of tabs and back slashes you included to no avail. Sure it's a simple answer. Appreciated it!

Golul-Madhu, 2024-10-24:
Have you been able to send large files to blob storage, if so could you provide the code.