espressif/esp-azure #24: iot hub device do not detect internet connection loss

anfelbeghoura, 2019-02-07:
Hey 
I'm working on ESP32, I used a modified version of the code found in `iothub_client_sample_mqtt.c` exemple to connect it with the Azure IotHub.

When I enabled the keep_alive in the file tlsio_openssl_compact.c for ESP32:
 https://github.com/Azure/azure-iot-pal-esp32/blob/master/pal/src/tlsio_openssl_compact.c 

OP_Option->keep_alive = 1;     // < 0 for system defaults, >= 0 to use supplied keep_alive, idle, interval, and count
OP_Option->keep_idle = 20;      // seconds before first keepalive packet (ignored if keep_alive <= 0)
OP_Option->keep_interval = 2;  // seconds between keepalive packets (ignored if keep_alive <= 0)
OP_Option->keep_count = 3;
 
to use it in the function:
SOCKET_ASYNC_HANDLE sock = socket_async_create(host_ipV4_address, tls_io_instance->port, false, OP_Option);
The ESP32 crashes : the Core 0 or 1 panics.
I got this error messages :
 
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400f74d8  PS      : 0x00060330  A0      : 0x800f771b  A1      : 0x3ffe6150
0x400f74d8: dowork_poll_dns at "PATH"/c
omponents/azure_iot/pal/src/tlsio_openssl_compact.c:603
 
A2      : 0x3ffe3324  A3      : 0xfb0f5f0d  A4      : 0xfffffc01  A5      : 0x000003ff
A6      : 0x7ff00000  A7      : 0x40590000  A8      : 0x3ffcd8c8  A9      : 0x00000000
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffe6684  A13     : 0x3ffe6684
A14     : 0x3ffe6684  A15     : 0x400f56d0  SAR     : 0x00000010  EXCCAUSE: 0x0000001d
0x400f56d0: onIoError at "PATH"compone
nts/azure_iot/azure/umqtt/src/mqtt_client.c:36
 
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
 
Backtrace: 0x400f74d8:0x3ffe6150 0x400f7718:0x3ffe6170 0x400f9227:0x3ffe6190 0x400f6318:0x3ffe61b0 0x400f504d:0x3ffe61e0 0x400f2d25:0x3ffe6220
0x400f1029:0x3ffe6240 0x400f082d:0x3ffe6270 0x400d7937:0x3ffe6290 0x400d7978:0x3ffe62c0 0x400918f1:0x3ffe62e0
0x400f74d8: dowork_poll_dns at "PATH"/c
omponents/azure_iot/pal/src/tlsio_openssl_compact.c:603
 
0x400f7718: tlsio_openssl_dowork at "PATH"/components/azure_iot/pal/src/tlsio_openssl_compact.c:729
 
0x400f9227: xio_dowork at "PATH"/compon
ents/azure_iot/azure/c-utility/src/xio.c:165
 
0x400f6318: mqtt_client_dowork at "PATH"/components/azure_iot/azure/umqtt/src/mqtt_client.c:1253
 
0x400f504d: IoTHubTransport_MQTT_Common_DoWork at "PATH"/components/azure_iot/azure/iothub_client/src/iothubtransport_mqtt_common.c:2965
 
0x400f2d25: IoTHubTransportMqtt_DoWork at "PATH"/components/azure_iot/azure/iothub_client/src/iothubtransportmqtt.c:115
 
0x400f1029: IoTHubClientCore_LL_DoWork at "PATH"/components/azure_iot/azure/iothub_client/src/iothub_client_core_ll.c:2154
 
0x400f082d: IoTHubClient_LL_DoWork at "PATH"/components/azure_iot/azure/iothub_client/src/iothub_client_ll.c:98
 
0x400d7937: iothub_client_sample_mqtt_run at "PATH"/main/iothub_client_sample_mqtt.c:499
 
0x400d7978: mqtt_sample at "PATH"/main/
iothub_client_sample_mqtt.c:531
 
0x400918f1: vPortTaskWrapper at "PATH"/esp-idf/components/freertos/port.c:401
 
With ESP8266 We can find this functionality and it works fine ; but I think it is not integrated yet to the ESP32.

Thank you

jitin17, 2019-02-12:
@anfelbeghoura I am worried that we no longer support [Azure/azure-iot-pal-esp32](https://github.com/Azure/azure-iot-pal-esp32). Can you try examples from this repository?

anfelbeghoura, 2019-02-12:
Well, in your exemple, the device doesn't identify the internet connection loss at all till the next mqtt client ping which is not the ideal behaviour.

jitin17, 2019-02-13:
@anfelbeghoura It is indeed a shortcoming in our example that it doesn't identify internet connection loss.  I am looking forward to adding this feature in the next few days. Thanks

jitin17, 2019-02-15:
@anfelbeghoura [azure-iot-sdk-c](https://github.com/Azure/azure-iot-sdk-c/blob/master/iothub_client/src/iothubtransport_mqtt_common.c) by default provides support for detecting internet connection loss. Read more about it [here](https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-reliability-features-in-sdks#connection-and-retry). By default, the connection policy is `IOTHUB_CLIENT_RETRY_EXPONENTIAL_BACKOFF`. But there is a bug in `azure-iot-sdk-c` which causes crash on connection retry. We have found a solution for it, and are planning to raise a PR soon. Thanks.

anfelbeghoura, 2019-02-18:
Thank you.
I'm looking forward to getting your feedback as soon as possible.

jitin17, 2019-02-21:
@anfelbeghoura We have added support for detecting internet connection loss in `iothub_client_sample_mqtt` example. Please try it out and let us know.

anfelbeghoura, 2019-02-24:
Thank you for the quick response, I will see the new version of iothub_client_sample_mqtt example.
 I actually made some changes in the SSL libraries and I used them to integrate the internet connection loss detection functionality to my system, I would like to know where did you make the changes please so I can understand what is going on.
Thank you

jitin17, 2019-03-01:
@anfelbeghoura Azure IoT SDK supports retry connection if the internet connection or connection to the wifi is lost. Read further [here](https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-reliability-features-in-sdks#connection-and-retry). By default, retry policy is `IOTHUB_CLIENT_RETRY_EXPONENTIAL_BACKOFF`. If you wish to add any other retry policy, then you can use `IoTHubClient_LL_SetRetryPolicy` API in the application.

closed 2019-03-01 by mahavirj