espressif/esp-azure #25: errors

naznaaa, 2019-02-11:
Hi,
while uploading the code in esp32 an error is encountered,error is specified below
C:/msys32/home/Nazna/esp/esp-azure-master/examples/prov_dev_client_ll_sample/main/custom_hsm_x509.c:8:29: fatal error: hsm_client_data.h: No such file or directory
similarly on compiling, errors are shown such that no such file or directory.can you please help

thanks

naznaaa, 2019-02-12:
Hi,
can you please explain the right way to run the device method code in esp32 using espidf(v3.1.2).
thanks

jitin17, 2019-02-12:
@naznaaa Can you send us the commit ID of head commit, and can you try updating the submodule and see if this problem gets resolved?

naznaaa, 2019-02-12:
Hi,
commit id -fb8d4fe307f15972785e78e94762dce7ddd393f7.

jitin17, 2019-02-12:
@naznaaa Thanks. I tried reproducing the issue, but it is not seen on my end. So, can you try updating the submodules with this command `git submodule update --init --recursive`?

naznaaa, 2019-02-12:
Hi,
can you please explain the way you build

mahavirj, 2019-02-14:
@naznaaa 

Either you can use `make` or `cmake` to build application. Following instructions can be used:
```
cd examples/iothub_client_sample_mqtt
make defconfig
make
```

Or

```
cd examples/iothub_client_sample_mqtt
idf.py build
```

Please note that on ESP-IDF side `release/v3.2` branch needs to be checked out.

naznaaa, 2019-02-14:
Hi,
I am working on espidf(v3.1.2 in windows.I just cloned the esp-azure-master folder to my esp folder and just build the example codes in it.while building errors i mentioned above like no such file or directory is seen .is it the right way to build in espidf? can yo please explain the steps to build
Thanks

mahavirj, 2019-02-14:
@naznaaa 

Get ESP-IDF using,
```
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout -b release/v3.2 origin/release/v3.2
export IDF_PATH=</path/to/esp-idf/>
```

Get `esp-azure` using,
```
git clone --recursive https://github.com/espressif/esp-azure.git
```

Then build example using,

```
cd esp-azure/examples/iothub_client_sample_mqtt
make defconfig
make
```

Let me know if you run into any issues with above.

```

naznaaa, 2019-02-15:
Hi,
still i have an error : C:/msys32/home/Nazna/esp/esp-idf/components/app_trace/app_trace.c:162:17: fatal error: eri.h: No such file or directory
compilation terminated.
on running the sample esp-azure/examples/iothub_devicetwin_samples_and_methods
can you please help to resolve it
Thanks

mahavirj, 2019-02-15:
@naznaaa Can you please try to build ESP-IDF provided example first, e.g. `esp-idf/examples/get-started/hello_world/`, just to ensure that at-least IDF setup is complete?

naznaaa, 2019-02-15:
Hi,
it is build but on monitoring it's shown that :0x400858b8: vPortCPUAcquireMutexIntsDisabledInternal at C:/msys32/home/Nazna/esp/esp-idf/components/freertos/port.c:403
 (inlined by) vPortCPUAcquireMutexIntsDisabled at C:/msys32/home/Nazna/esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vPortCPUAcquireMutex at C:/msys32/home/Nazna/esp/esp-idf/components/freertos/port.c:394
hello world is not displayed

mahavirj, 2019-02-15:
@naznaaa I am running out of ideas, I see no reason why basic hello-world example from IDF itself is problematic at your end, can you please confirm instruction and hardware from https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html?

naznaaa, 2019-02-15:
Hi,
hello world is working ,still on on running the sample
 azure/examples/iothub_devicetwin_samples_and_methods got the  error : C:/msys32/home/Nazna/esp/esp-idf/components/app_trace/app_trace.c:162:17: fatal error: eri.h: No such file or directory
compilation terminated
can you please help

naznaaa, 2019-02-22:
Hi,
now the sample prov_dev_client_ll_sample is able to bulid but on monitoring an error is rendered:
E (23638) esp-tls: mbedtls_pk_parse_keyfile returned -0x3c00
can you please help
thanks

mahavirj, 2019-02-22:
@naznaaa Please ensure that you have correct device credentials in application as mentioned at https://github.com/espressif/esp-azure/tree/master/examples/prov_dev_client_ll_sample#device-leaf-certificate-and-key

naznaaa, 2019-02-25:
Hi,
I uploaded the certificate below,still i have the error:E (23638) esp-tls: mbedtls_pk_parse_keyfile returned -0x3c00 on running the sample prov_dev_client_ll_sample.can you please check and let me know

[certs.zip](https://github.com/espressif/esp-azure/files/2899477/certs.zip)

jitin17, 2019-03-01:
@naznaaa Follow the instructions given in the link to generate certificates and key: https://github.com/Azure/azure-iot-sdk-c/blob/master/tools/CACertificates/CACertificateOverview.md.
Once you are done with this please create an enrollment in azure IoT cloud's DPS service.

naznaaa, 2019-03-07:
Hi,
Among these examples which one is for Device Provisioning - Provision Device with Group Enrollment 
using X.509 certificate feature.
I couldn't verify these examples are for Group Enrollment or individual. Please provide an example for Provisioning device using Group Enrollment

naznaaa, 2019-03-11:
Hi,
can you please provide sample for direct method using auto provisioning(using x.509 authentication) instead of connecting device to iot hub through connection string.please reply soon

mahavirj, 2019-04-17:
@naznaaa This issue has diverted from original build error. I will presume that original build issue is fixed and close this. Please feel free to reopen in case issue exists.

closed 2019-04-17 by mahavirj