espressif/esp-azure #26: ESP32 Azure-IoT Crash Cases

soham-nandi, 2019-02-12:
Hello,
I have been running the Azure IoT example-code on my ESP32 and checking when and why the ESP32 crashes. I'm getting the following error message on my Serial Monitor - 

Info: >>>Re-connect.

Info: Initializing SNTP

assertion "Operating mode must not be set while SNTP client is running" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/apps/sntp/sntp.c", line 600, function: sntp_setoperatingmode
abort() was called at PC 0x400e7e33 on core 1

Backtrace: 0x4008c828:0x3ffb1b80 0x4008ca59:0x3ffb1ba0 0x400e7e33:0x3ffb1bc0 0x40120381:0x3ffb1bf0 0x400d6cb6:0x3ffb1c10 0x400d6d12:0x3ffb1c30 0x400d5ce6:0x3ffb1c50 0x400d601a:0x3ffb1c90 0x400d6181:0x3ffb1cb0 0x400d2fa9:0x3ffb1cd0 0x400e2599:0x3ffb1fb0 0x4008eaa9:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4

Now, what is the meaning of the "assertion" and "backtrace" messages?
More importantly, what are the causes for the ESP32 to reboot in this case?

Kind regards,
Soham

mahavirj, 2019-02-14:
@soham-nandi Which example did you use from `esp-azure` repository for testing purpose?

mahavirj, 2019-02-20:
Closing, please feel free to reopen new issue in case you observe any issue with examples in this repository.

closed 2019-02-20 by mahavirj

jpagliaccio, 2019-09-11:
Using the Microsoft examples I am having the same issue. 

08:53:42.059 -> Send Confirmation Callback finished.
08:53:54.969 -> Info: >>>Connection status: timeout
08:53:59.978 -> Info: >>>Re-connect.
08:53:59.978 -> Info: Initializing SNTP
08:53:59.978 -> assertion "Operating mode must not be set while SNTP client is running" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/apps/sntp/sntp.c", line 600, function: sntp_setoperatingmode
08:54:00.011 -> abort() was called at PC 0x400e43af on core 1
08:54:00.011 -> 
08:54:00.011 -> Backtrace: 0x4008cbcc:0x3ffb1be0 0x4008cdfd:0x3ffb1c00 0x400e43af:0x3ffb1c20 0x4011babd:0x3ffb1c50 0x400d3b3e:0x3ffb1c70 0x400d3e9e:0x3ffb1c90 0x400d336a:0x3ffb1cb0 0x400d369e:0x3ffb1cf0 0x400d3805:0x3ffb1d10 0x400d25ce:0x3ffb1d30 0x400dfd2d:0x3ffb1fb0 0x40088b69:0x3ffb1fd0
08:54:00.011 -> 
08:54:00.011 -> Rebooting...
08:54:00.044 -> ets Jun  8 2016 00:22:57
08:54:00.044 -> 
08:54:00.044 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
08:54:00.044 -> configsip: 0, SPIWP:0xee
08:54:00.044 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
08:54:00.044 -> mode:DIO, clock div:1
08:54:00.044 -> load:0x3fff0018,len:4
08:54:00.044 -> load:0x3fff001c,len:928
08:54:00.044 -> ho 0 tail 12 room 4
08:54:00.044 -> load:0x40078000,len:8424
08:54:00.044 -> ho 0 tail 12 room 4
08:54:00.044 -> load:0x40080400,len:5868
08:54:00.044 -> entry 0x4008069c
08:54:00.376 -> ESP32 Device
08:54:00.376 -> Initializing...
08:54:00.376 ->  > WiFi
08:54:00.376 -> Connecting...
08:54:01.006 -> ..WiFi connected
08:54:01.504 -> IP address: 192.168.1.71
08:54:01.504 ->  > IoT Hub
08:54:01.504 -> Info: Initializing SNTP
08:54:02.499 -> Info: SNTP initialization complete
08:54:02.499 -> Info: IoT Hub SDK for C, version 1.1.23
08:54:05.285 -> Info: >>>Connection status: connected