espressif/esp-azure #27: ESP32 DNS Lookup failure causing ESP32 to Crash

soham-nandi, 2019-02-12:
Hello,

In the example-code for the ESP32 Azure-IoT code, I keep getting an error as follows -

Info: SNTP initialization complete

Info: IoT Hub SDK for C, version 1.1.23

Info: Failed DNS lookup

This error causes my ESP32 to restart entirely. What would be all the possible cases for that DNS lookup error?

Kind regards,
Soham

jitin17, 2019-02-13:
@soham-nandi Can you tell us which example are you using?

soham-nandi, 2019-02-14:
In the Arduino IDE examples, there's the "getting started" example for "ESP32 Azure IoT Arduino". On running that code, I'm getting that error. I haven't edited the code much, yet there seems to be this problem.

mahavirj, 2019-02-14:
@soham-nandi You might want to file issue in project from where you downloaded `ESP32 Azure IoT Arduino` example code in this case. Interim if you run into any issues with examples/application from this repository, please feel free to raise issue.

closed 2019-02-20 by mahavirj