espressif/esp-azure #29: how can we come out of azure function

naznaaa, 2019-04-15:
Hi,

we have to send some messages to azure iot ,after that we have to come out of this functon to do some other action
we use the sample https://github.com/espressif/esp-azure/tree/master/examples/iothub_client_sample_mqtt 
here after sending messages and recieving confirmation we have to do some other function, but here it's going with ->pingreq and ->pingresp 
how can we come out of this fuction can you please help

jitin17, 2019-04-17:
@naznaaa `SendConfirmationCallback ` is the confirmation callback function in this  https://github.com/espressif/esp-azure/tree/master/examples/iothub_client_sample_mqtt example. So, invoke the function you want to call after receiving confirmation in `SendConfirmationCallback` function.

mahavirj, 2019-04-30:
Closing, assuming above answers the question. Please feel free to reopen if there is any issue.

closed 2019-04-30 by mahavirj