espressif/esp-azure #33: tlsio_esp_tls_send_async without a prior successful open

fodgeri, 2019-05-13:
Hi!

I have a problem to maintain a stable connection to the Azure IoT Hub with the esp-azure sdk. I would like to upload the data from a queue to the hub continuously but the connection fails after 1 or 2 hours and the esp could not reconnect. This is the error message I get:

IoTHubClient_LL_SendEventAsync accepted message [1741] for transmission to IoT Hub.
Error: Time:Sat May  4 08:31:18 2019 File:/home/dev/esp/esp-azure/port/src/tlsio_esp_tls.c Func:tlsio_esp_tls_send_async Line:543 tlsio_esp_tls_send_async without a prior successful open
Error: Time:Sat May  4 08:31:18 2019 File:/home/dev/esp/esp-azure/azure-iot-sdk-c/umqtt/src/mqtt_client.c Func:sendPacketItem Line:371 542: Failure sending control packet data
Error: Time:Sat May  4 08:31:18 2019 File:/home/dev/esp/esp-azure/azure-iot-sdk-c/umqtt/src/mqtt_client.c Func:mqtt_client_publish Line:1059 Error: mqtt_client_publish send failed
Error: Time:Sat May  4 08:31:18 2019 File:/home/dev/esp/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c Func:publish_mqtt_telemetry_msg Line:865 Failed attempting to publish mqtt message

Here is my code for the task:

```
void azure_task(void *pvParameter)
{
    xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT,
                        false, true, portMAX_DELAY);
    ESP_LOGI(TAG, "Connected to AP success!");

    (void)IoTHub_Init();

    int result;
    esp_err_t err;
    IOTHUB_SECURITY_TYPE security_type;
    //security_type = IOTHUB_SECURITY_TYPE_SAS;
    security_type = IOTHUB_SECURITY_TYPE_X509;

    (void)iothub_security_init(security_type);

    IOTHUB_DEVICE_CLIENT_LL_HANDLE device_ll_handle;
 
    if ((device_ll_handle = IoTHubDeviceClient_LL_CreateFromConnectionString(conn_string, MQTT_Protocol)) == NULL)
    {
        (void)printf("Failure creating device Auth!\r\n");
        result = __LINE__;
    }
    else
    {
        size_t msg_count = 0;
        IOTHUB_CLIENT_SAMPLE_INFO iothub_info;

        iothub_info.stop_running = 0;

        (void)IoTHubDeviceClient_LL_SetMessageCallback(device_ll_handle, receive_msg_callback, &iothub_info);
        (void)IoTHubDeviceClient_LL_SetConnectionStatusCallback(device_ll_handle, connection_status_callback, &iothub_info);
        
        while(1) 
        {
            while (xQueueReceive(azure_queue, &dataToAzure, portMAX_DELAY) == pdTRUE)
            {
                    
                    static char msgText[1024];
                    sprintf_s(msgText, sizeof(msgText), "%s", dataToAzure);
                    IOTHUB_MESSAGE_HANDLE msg_handle = IoTHubMessage_CreateFromByteArray((const unsigned char*)msgText, strlen(msgText));
                    if (msg_handle == NULL)
                    {
                        (void)printf("ERROR: iotHubMessageHandle is NULL!\r\n");
                    }
                    else
                    {
                        IoTHubDeviceClient_LL_SendEventAsync(device_ll_handle, msg_handle, NULL, NULL);
                        (void)printf("IoTHubClient_LL_SendEventAsync accepted message [%zu] for transmission to IoT Hub.\r\n", msg_count);
                        msg_count++;
                        IOTHUB_CLIENT_STATUS status;

                        while ((IoTHubClient_LL_GetSendStatus(device_ll_handle, &status) == IOTHUB_CLIENT_OK) && (status == IOTHUB_CLIENT_SEND_STATUS_BUSY))
                        {
                            IoTHubClient_LL_DoWork(device_ll_handle);
                            vTaskDelay(100);
                        }
                        IoTHubMessage_Destroy(msg_handle);
                    }
            }

        }
    }

    // Clean up the iothub sdk handle
    IoTHubDeviceClient_LL_Destroy(device_ll_handle);

    // Free all the sdk subsystem
    (void)IoTHub_Deinit();

    vTaskDelete(NULL);
}
```

BR,
Geri

drwill-ms, 2019-11-07:
Hi Geri -

I got a very similar error message that you did and found your issue while searching for a solution to mine. In my case, however, the failure happens immediately.

My error looked something like:

```
/<redacted>/components/esp-azure/port/src/tlsio_esp_tls.c
Func:tlsio_esp_tls_send_async Line:551 tlsio_esp_tls_send_async without a prior successful open
Error: Time:Wed Nov  6 19:13:43 2019 File:<redacted>/components/esp-azure/azure-iot-sdk-c/umqtt/src/mqtt_client.c
Func:sendPacketItem Line:371 550: Failure sending control packet data
Error: Time:Wed Nov  6 19:13:43 2019 File:<redacted>/components/esp-azure/azure-iot-sdk-c/umqtt/src/mqtt_client.c Func:mqtt_client_publish Line:1059 Error: mqtt_client_publish send failed
Error: Time:Wed Nov  6 19:13:43 2019 File:<redacted>/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c
Func:publish_mqtt_telemetry_msg Line:865 Failed attempting to publish mqtt message
```

I was able to track it down to a new function call I added:

```C
IOTHUB_MESSAGE_RESULT contentTypeResult = IoTHubMessage_SetContentTypeSystemProperty(heapMessageHandle, "application/json");
```

It turns out the slash in application/json was damaging the URL, and the SDK does not automatically URL escape these values. To fix that, I added this snippet to my device client initialization, and it fixed it.

```C
bool encodeDecode = true;
IOTHUB_CLIENT_RESULT setOptionResult = IoTHubDeviceClient_LL_SetOption(
    deviceClientHandle,
    OPTION_AUTO_URL_ENCODE_DECODE,
    &encodeDecode);
```

So for me, this error was misleading. However, I don't see that you are adding any content,m and yours works for a while, so that would make my solution not apply to your situation.

I notice you are using certificates, and this is happening after a couple of hours. Perhaps you are running out of stack or heap, and when auth needs to be challenged again, there isn't enough to perform the operation. Just a thought. You might try calling the `esp_get_free_heap_size()` and `esp_get_minimum_free_heap_size()` functions in your event loop to keep track of available heap, and see if that is going down over time. I looked over your code and don't see any obvious leaks, but I've been out of the C game for over a decade and only recently getting back into it for MCU development.

Best of luck,
David

fodgeri, 2019-11-12:
Hi David,

thank you for your answer, I had to change my architecture to avoid this problem. Right now I'm using a raspberry pi "hub" for these continuous uploads but I'll definitely try your suggestions and give the esp32 an other chance.

Thank you again,
Geri

JayEnaR, 2020-07-19:
I too get the same error 
In my case I send a message every minute or so, and after about 10 messages sent I get the error:

E (953737) esp-tls: read error :-76:
File: ../components/esp-azure/port/src/tlsio_esp_tls.c Func:tlsio_esp_tls_send_async 
Line:552 tlsio_esp_tls_send_async without a prior successful open

sorrowed, 2021-10-04:
I am also seeing this same error:
`
E (428064) esp-tls-mbedtls: read error :-76:
Error: Time:Mon Oct  4 10:31:25 2021 File:../Libraries/espressif-azure-sdk/port/src/tlsio_esp_tls.c Func:tlsio_esp_tls_send_async Line:552 tlsio_esp_tls_send_async without a prior successful open
Error: Time:Mon Oct  4 10:31:25 2021 File:../Libraries/espressif-azure-sdk/azure-iot-sdk-c/umqtt/src/mqtt_client.c Func:sendPacketItem Line:395 Failure sending control packet data
Error: Time:Mon Oct  4 10:31:25 2021 File:../Libraries/espressif-azure-sdk/azure-iot-sdk-c/umqtt/src/mqtt_client.c Func:mqtt_client_publish Line:1117 Error: mqtt_client_publish send failed
Error: Time:Mon Oct  4 10:31:25 2021 File:../Libraries/espressif-azure-sdk/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c Func:publish_mqtt_telemetry_msg Line:900 Failed attempting to publish mqtt message
`
Using IDF v4.2.2 on the ESP32

Any progress/info on how to solve this issue?