espressif/esp-azure #34: Use a newer version of the Azure-iot-sdk-c

Almagnus1, 2019-05-15:
The version of the Azure IoT C SDK used is over three months old, while the latest release was last month ( https://github.com/Azure/azure-iot-sdk-c/releases/tag/2019-04-11 ).

Any chance that esp-azure can be updated to use a more current version of the Azure C SDK?

mahavirj, 2019-05-16:
@Almagnus1 Any specific feature and/or fix you would like from latest Azure IoT release?

naznaaa, 2019-06-19:
On using latest release i am getting an error :C:/msys32/home/Nazna/esp/SMART_WARMER/components/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c:1518:32: error: 'request_id' may be used uninitialized in this function [-Werror=maybe-uninitialized]
I am using idf master branch in windows
can you please help

Almagnus1, 2019-06-19:
I'd like to see https://github.com/Azure/azure-iot-sdk-c/tree/master/provisioning_client/samples/prov_dev_client_ll_sample brought over to the ESP-IDF dev environment.

shahpiyushv, 2020-07-02:
@Almagnus1 , closing this issue as the prov_dev_client_ll_sample was brought in long back and azure-iot-c-sdk has been updated to the most recent pre-release LTS_06_2020_Ref01.

closed 2020-07-02 by shahpiyushv