espressif/esp-azure #35: Prov_Device_LL_Register_Device infinite loop

fodgeri, 2019-05-18:
Hi!

I have a provisioning issue with the latest esp-azure sdk. I'm using the prov_dev_client_ll_sample program and problem is at the Prov_Device_LL_Register_Device part:

```
if (Prov_Device_LL_Register_Device(handle, register_device_callback, &user_ctx, registation_status_callback, &user_ctx) != PROV_DEVICE_RESULT_OK)
{
    (void)printf("failed calling Prov_Device_LL_Register_Device\r\n");
}
else
{
    do
    {
        Prov_Device_LL_DoWork(handle);
        ThreadAPI_Sleep(user_ctx.sleep_time);
    } while (user_ctx.registration_complete == 0);
}

```
Here the provisioning stays in the loop and never exits, not calling the register_device_callback function. Here is the console output I get:

```
I (6288) tcpip_adapter: sta ip: 192.168.0.42, mask: 255.255.255.0, gw: 192.168.0.1
I (6288) azure: Connected to AP success!
Initializing SNTP
ESP platform sntp inited!
Time is not set yet. Connecting to WiFi and getting time over NTP. timeinfo.tm_year:70
Waiting for system time to be set... tm_year:0[times:1]
I (8308) platform: The current date/time is: Sat May 18 07:52:20 2019
Provisioning API Version: 1.2.14
Iothub API Version: 1.2.14
Provisioning Status: PROV_DEVICE_REG_STATUS_CONNECTED
Provisioning Status: PROV_DEVICE_REG_STATUS_ASSIGNING
Provisioning Status: PROV_DEVICE_REG_STATUS_ASSIGNING
```

This sample program worked a month ago for sure.

BR,
Geri

jitin17, 2019-06-19:
@fodgeri I think your issue might be similar to https://github.com/espressif/esp-azure/issues/37, which has been fixed now. So, can you confirm whether you are still facing the problem?

shahpiyushv, 2019-10-23:
@fodgeri , did you try the above solution?

fodgeri, 2019-10-23:
Yes, it's working now! Thank You!

closed 2019-10-23 by fodgeri