espressif/esp-azure #36: ESP32-DevKitC : "Operating mode must not be set while SNTP client is running" issue.

JonS9999, 2019-05-20:
Hi all.

I've just been playing with getting an ESP32-DevKitC working with Azure, and it works for the most part, but I'm seeing the "Operating mode must not be set while SNTP client is running" error that other people have reported (e.g., issue # 26).

I started with the SimpleMQTT example to verify the connection, and that worked, however I've rewritten it so can handle WiFi and Azure disconnects and reconnects.  I did this by creating a state machine that does things like (1) wait for the WiFi link to come up, then (2) once it does, attempt to connect to Azure, then (3) once it's connected go into a "CONNECTED" state so we can send messages back and forth, etc...

This works well *IF* the WiFi is up before my ESP32 is up -- the ESP32s find and connects to the WiFi, finds and connects to Azure, is able to send and receive messages to/from Azure, etc...  Life is good.

However, if the WiFi drops and comes back, after the ESP has reconnected to the WiFi, when the ESP32 reconnects to Azure, I get the dreaded SNTP related error, and the ESP32 reboots.

I saw on another post that the ESP32<->Azure connection is closed if there is inactivity for a given amount of time (the OP of that thread saw the same SNTP error when his IoT device tried to reconnect to Azure after inactivity).  Note that there were also some "fixes" proposed, but this required the OP to edit the libraries (if I remember correctly).  I've seen this problem as well since my IoT only sends a message to Azure when something changes -- if I let my ESP32 run long enough (without me flipping switches, etc), I see the reconnect/SNTP issue.

I was wondering if there is an official fix for this problem?

Thank you for your time.

JonS9999, 2019-05-20:
I forgot to mention that when the code detects that the WiFI has dropped, it will execute an Esp32MQTTClient_Close() call, but that errors out since there is no active WiFi/Azure connection.

JonS9999, 2019-05-25:
No responses??  :-?  :-(

jitin17, 2019-05-28:
Are you using Arduino based Azure library? 
I don't find `Esp32MQTTClient_Close()` API in this repository.

JonS9999, 2019-05-28:
I thought I was, but honestly, I'm not 100% sure.  In my attempts to get my ESP32 working with Azure, I tried a number of different libraries until I found one that worked.  According to the Arduino IDE, I have the following Azure related libraries installed:

   - AzureIoTHub (by Microsoft, ver 1.0.45)
   - AzureIoTHubMQTTClient (by Andri Yadi, ver 0.2.3)
   - AzureIoTProtocol_MQTT (by Microsoft, ver 1.0.45)
   - AzureIoTUtility (by Microsoft, ver 1.0.45)

Under the File -> Examples menu, I started with ESP32 Azure IoT Arduino -> SimpleMQTT in order to get something working.  Once I had this up and running, I wrote my own, based off of the SimpleMQTT example.  This is the example that uses "Esp32MQTTClient.h" and the corresponding Esp32MQTTCLient_xxxx() calls.  It doesn't use _Close directly, but it's in the include file, so I thought I'd give it a try.

Under the File -> Examples -> INCOMPATIBLE menu, there are several Azure IoT example -- I did not try any of these.

jitin17, 2019-05-28:
@JonS9999 Can you try the samples given at examples directory of this repository?

JonS9999, 2019-05-28:
FYI, I was able to fix the problem based on the information I found here:

   https://github.com/VSChina/ESP32_AzureIoT_Arduino/issues/5

On Jul 19, 2018, lirenhe suggested adding the following code to SNTP_Init() in sntp_lwip.c :

   if ( sntp_enabled() ) {
       sntp_stop();
   }

This did fix the problem.  Basically, SNTP_Init() would fail the second time it was called.  The above code gracefully terminates/cleans up the existing SNTP code which allows it to be restarted successfully.

I merged that code into this file :

C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\AzureIoT\src\az_iot\c-utility\pal\lwip\sntp_lwip.c

I will try your examples to see if I have any better luck (without needing to modify any library code).

Thank you for your time.  :-)

-Jon

mahavirj, 2019-06-19:
@JonS9999 I will close this issue as problem is fixed. If you still find any problem, please feel free to raise new issue.

closed 2019-06-19 by mahavirj