espressif/esp-azure #37: MQTT message length issue

KimBoendergaard, 2019-05-24:
We have a problem receiving MQTT packages longer than approximately 280 bytes.
Any one having seen a similar problem?

nrmka, 2019-06-03:
Hi @KimBoendergaard,

I have a similar issue. If complete twin ("desired" + "reported") is longer than 287 bytes, esp32 hangs on receiving it.
In my application I create task with connection to the cloud and updating twin. However, for some messages callback on twin update is never called.

I use low level functions:

```
void twinUpdateCallback(int statusCode, void* userContext)
{
    (void)userContext;
    printf("Twin update status: %d\r\n", statusCode);
    m_curMsgTxInProgress = false;
}

...
m_curMsgTxInProgress = true;
IoTHubDeviceClient_LL_SendReportedState(m_deviceHandle, (const unsigned char*)msg, strlen(msg), twinUpdateCallback, nullptr);
printf("Wait sent\n");
do
{
    IoTHubDeviceClient_LL_DoWork(m_deviceHandle);
    ThreadAPI_Sleep(100);
} while (m_curMsgTxInProgress);
```

The application sends my message to the cloud and I can see it there. But the app cannot leave `do while`.

I started logging in azure sdk and print buffer in mqtt_codec_bytesReceived() from mqtt_code.c. Here is [log.txt](https://github.com/espressif/esp-azure/files/3247664/log.txt).

nrmka, 2019-06-03:
I tried this https://github.com/espressif/esp-azure/pull/39 and it helped me.

KimBoendergaard, 2019-06-14:
I fully agree. #39 fixes the problem. Let's hope the pull request gets merged in

mahavirj, 2019-06-19:
@KimBoendergaard @nrmka Sorry took us while for merging PR. Fixed with https://github.com/espressif/esp-azure/commit/5f421bbd7ef8e6f5502204e9cd7a5f494070b39a

closed 2019-06-19 by mahavirj