espressif/esp-azure #38: Unable to port provisioning example

DumDumin, 2019-05-26:
Hey guys,

I am working on Tag v1.0 with an ESP32 chip.
I managed to expand your **prov_dev_client_ll_sample** to use symetric keys by adding the **HSM_TYPE_SYMM_KEY** flag to the **component_compile_definitions** and adding the **DUSE_PROV_MODULE** flags in the **port/CMakeLists.txt**:

```
component_compile_definitions (
    USE_LWIP_SOCKET_FOR_AZURE_IOT
	# HSM_TYPE_X509
	HSM_TYPE_SYMM_KEY
	HSM_TYPE_SAS_TOKEN
)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -DUSE_PROV_MODULE")
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -DUSE_PROV_MODULE")
```

 and creating a **custom_hsm_symkey.c** file.

So far so good, everything is working and the esp32 is correctly provisioned and connects to the right iot hub afterwards :)

Anyway, if I try to use the provisioning sdk in my own project code I get the following linker error: 

```
port/libport.a(http_proxy_io.c.obj):(.literal.http_proxy_io_create+0x2c): undefined reference to `socketio_get_interface_description'
port/libport.a(http_proxy_io.c.obj): In function `http_proxy_io_create':
C:\Source\tyc-esp32-iot\build/../components/esp-azure/azure-iot-sdk-c/c-utility/src/http_proxy_io.c:140: undefined reference to `socketio_get_interface_description'
```

My main CMakeLists.txt already includes this, like in the example:

`set (EXTRA_COMPONENT_DIRS "${CMAKE_CURRENT_LIST_DIR}/components/esp-azure/port")`

Any suggestions what I might missing in the build process?
Thank you!

DumDumin, 2019-05-27:
I found the issue. Your example code in `prov_dev_client_ll_sample.c` has the following implementation to solve the problem:

`const IO_INTERFACE_DESCRIPTION* socketio_get_interface_description(void)
{
    return NULL;
}`

I would suggest to put this code in a new file in the port folder to avoid trouble using your library.

sagar448, 2019-07-26:
Hey DumDumin, would you be able to help? I cannot get the prov_dev_client_11_sample.c to work. It always says undefined reference to `_binary_leaf_private_key_pem_start'. Please help!

chinmaychhajed, 2019-09-25:
Hello @sagar448, 

This issue arises when you use `CMake` with provisioning example. You can instead use `make` for this example. It should build without errors.

chinmaychhajed, 2019-09-25:
You can try applying [this](https://github.com/espressif/esp-azure/files/3651303/cmake_cert_changes.patch.txt) patch. Then it should work with `CMake` also. Make sure you are using idf version 3.x

ryanwinter, 2020-02-20:
Just hit this problem, thankyou for this thread with the solution! 

Disappointed that this is 9 months on and a fix still hasn't been rolled into the code.

shahpiyushv, 2020-06-04:
Cmake support was already fixed way back in October by [this commit](https://github.com/espressif/esp-azure/commit/795e1a6ec6ef07666797a233407aa7ada71043f3). Closing this issue.

closed 2020-06-04 by shahpiyushv