espressif/esp-azure #41: az iot hub: 'monitor-events' is not in the 'az iot hub' command group. See 'az iot hub --help'.

pelikhan, 2019-06-21:
The command in the readme does not work with the latest az cli (2.0.67)

```
>az iot hub monitor-events -n "..." -login "..."
az iot hub: 'monitor-events' is not in the 'az iot hub' command group. See 'az iot hub --help'.
```

shahpiyushv, 2019-06-21:
@pelikhan , please execute `az extension add --name azure-cli-iot-ext` once and try again.

pelikhan, 2019-06-21:
It went a bit further but now it complains that the connection string I got from the Azure portal is missing SharedAccessKeyName.

```
IoT Hub connection string has missing property: SharedAccessKeyName
```

shahpiyushv, 2019-06-21:
@pelikhan , for which command did you get this error? Once you login using az login, for monitoring the events, `az iot hub monitor-events -n <Hub name>` should be sufficient. We will correct the steps accordingly.

pelikhan, 2019-06-21:
Dropping the token in the ``monitor-events`` invocation worked. thanks.

shahpiyushv, 2019-10-15:
@pelikhan , I am closing this issue as it has been resolved. We have also updated the documentation so that trying out the examples is easy.

closed 2019-10-15 by shahpiyushv