espressif/esp-azure #42: Missing section on building SDK

pelikhan, 2019-06-21:
Where are the rest of the step after https://github.com/espressif/esp-azure#1-cloning-git-submodules ? Do we need to install separately the Expressif compiler toolchain? What kind of layout? Please provide an end-to-end list of step to get this sdk to build.

(extra point: instructions that work on Windows, or specify which OS/distro is needed to make it work).

shahpiyushv, 2019-06-21:
@pelikhan , we will indeed update the README soon enough so that it would be easier to start the development process, and work with the latest Azure IoT hub. For the time being, please check the pointers under https://github.com/espressif/esp-azure#5-sdk to get the ESP IDF. The IDF README at https://github.com/espressif/esp-idf/blob/master/README.md will also help you getting started with development on ESP32. Once the ESP IDF is ready and configured, you can use the esp-azure repo and all its submodules to start working on Azure IoT.

pelikhan, 2019-06-25:
What is the best way to integrate the SDK into an existing ESP-ISF project?

JAHTKELD, 2019-07-18:
Do you find the soluttions? I also try to use this SDK with ESP32 module under Widnows but I can integrate this SDK to esp-idf.
Thanks

shahpiyushv, 2019-07-18:
In order to include this in an existing project, you can do the following:

- Copy the entire contents of esp-azure repo to esp-idf/components/esp-azure folder. Since esp-azure already has the component.mk file, this will be treated as any other component
- Check any of the examples azure_main.c file. You will see that most code is only to establish the Wi-Fi connection. The "azure_task" waits for a successful connection, and once a connection is established, the functions  iothub_client_sample_mqtt_run() or iothub_client_device_twin_init() are used.

You can follow a similar flow in your application. Let me know if this helps or if you are looking for something else.

shahpiyushv, 2020-07-02:
Closing this issue as appropriate build instructions have been added in the docs and the latest esp-azure is compatible even with cmake.

closed 2020-07-02 by shahpiyushv