espressif/esp-azure #43: Implement symmetric key method on ESP32

clementcres, 2019-06-24:
Hello,

I am trying to port the example `prov_dev_client_II_sample` to get the symmetric key method working. If I understand well, I don't need any certificates with this method so I removed `CFLAGS += -DSET_TRUSTED_CERT_IN_SAMPLES` in the component file.

But I still got the following error:
`Missing PEM file esp-azure/examples/prov_dev_client_ll_sample/main/certs/leaf_private_key.pem. This file identifies the ESP32 to Azure DPS for the example, see README for details.`
`make[1]: *** [esp-azure/examples/prov_dev_client_ll_sample/main/component.mk:15: esp-azure/examples/prov_dev_client_ll_sample/main/certs/leaf_private_key.pem] Error 1`
`make[1]: *** Attente des tâches non terminées....`
`make: *** [esp/esp-idf/make/project.mk:582: component-main-build] Error 2`
`make: *** Attente des tâches non terminées....`

Did I miss something?

Thank you,
Best regards
Clement

sagar448, 2019-07-26:
I also got this error, couldn't get around it.

Lisa999, 2020-07-10:
#74