espressif/esp-azure #46: Azure IoT Build Error MOCKABLE_FUNCTION on mac

frederic-thibault, 2019-07-10:
Hi,

I receive a lot of error like this one when I build the sample project on mac. Examples not using IoT Hub builds and works fine. I just changed my mac and it was working just fine in my old one.

/Users/fred/esp/esp-azure/port/../azure-iot-sdk-c/c-utility/inc/azure_c_shared_utility/strings.h:36:1: error: unknown type name 'IF'
 MOCKABLE_FUNCTION(, int, STRING_replace, STRING_HANDLE, handle, char, target, char, replace);
 ^~~~~~~~~~~~~~~~~

cheezum, 2019-08-06:
After updating I have the same problem.
Previous release was v4.0-dev-1381. Updating to v4.0-dev-1434 gives me now the same errors as described by frederic-thibault
My system is running on ubuntu 18.10

hfmartinez, 2019-08-31:
Hi @cheezum  
Current, I have the same problem, changed the path file and created a new project with the example configuration, but i can't build the solution. do you have any idea?. I compile in Windows. Thanks.

errors:
../strings.h:35:1: error: unknown type name 'IF'
 MOCKABLE_FUNCTION(, int, STRING_compare, STRING_HANDLE, s1, STRING_HANDLE, s2);

chinmaychhajed, 2019-09-01:
Hi,
Currently esp-azure does not work properly with IDF version 4 and CMake. esp-azure has been tested with IDF version 3.3 and Make build system. If you have already cloned the esp-idf master then use `$ cd $IDF_PATH; git checkout release/v3.3` to change to IDF version 3.3. Additionally you can follow provided steps to avoid some common problems:

-    update path of toolchain to xtensa version 5.2.0
-   Run `$ cd $IDF_PATH; git submodule update --init --recursive`
-    run `$ make -C $IDF_PATH/tools/kconfig clean`
-    `cd` to your project directory
-    run `$ make clean`
-    run `$ make -j8 defconfig`
-    then `$ make -j8 flash monitor`

    `$IDF_PATH` is path to esp-idf (version 3.3 in this case) directory

Please let me know after following these steps your project works or not.

cheezum, 2019-09-02:
Hi @hfmartinez 
What worked for me was turning on an option in the menuconfig:
Compatibility Options -> Include headers accross components as before IDF v4.0

However, I can't get the error back when turning this option off again.

Cheers.

hfmartinez, 2019-09-03:
Hi @chinmaychhajed 

It was currently possible to connect and send data to IoT Hub, with the change to version 3.3 and the steps you mentioned were solved, thank you.

shahpiyushv, 2019-10-15:
@frederic-thibault @hfmartinez Some minor changes have been made to esp_azure so that it can compile with ESP IDF tag: v4.0-beta1