espressif/esp-azure #47: How can I use this SDK?

JAHTKELD, 2019-07-18:
I have installed the ESP-IDF   using  ESP-IDF Tools Installer (Windows). I have a folder C:/esp-idf and I can compiler several examples of ESP32  but now i need the azure SDK and I don´t know what I have to do with this repositorie  Need I compiler for ESP32? There is a port folder but it not seems to have the neccesary files to work with azure. I am an microcontroller programmer and I use tools like AtmelStudio and other microcontroller compiler  and I not experiences in this kind of library and what I have to do for the compiler find the right files. What files or folder I need to put in esp-idf/component folder?
Thanks

shahpiyushv, 2019-07-18:
@JAHTKELD , since you can compile the ESP-IDF examples, I am assuming that you have the correct tool  chain and IDF_PATH set.

Now, in order to use esp-azure, follow these steps:

1. Clone this esp-azure repository to a suitable location
2. Enter esp-azure folder and execute `git submodule update --init --recursive`
3. Go to one of the example folders (say examples/iothub_client_sample_mqtt)
4. Compile this the same way as you do for other examples (viz. make deconfig;  make flash monitor)

Let me know if this helps or if you need some additional information.

JAHTKELD, 2019-07-18:
Hello Shahpiyushv: 

Thanks for your quick answer. I have tried it and I have this error

```
Component directory C:/esp-idf/components/esp-azure does not contain a CMakeLists.txt file. No component will be added
```

and a the end of compitation

```
 fatal error: iothub_client.h: No such
file or directory
 #include "iothub_client.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
[818/824] Building C object esp-idf/wi...ifi_provisioning.dir/src/manager.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
I am using the latest version of the 

**Getting Started Guide for the latest (master branch)  ESP-IDF,** 

and using idf.py comand for config and bluid 

`idf.py and menuconfig` 
`idf.py build`

JAHTKELD, 2019-07-18:
Hello:
I have also tried to copy only azure-iot-sdk-c folder (it has a cmakelists.txt file) but I have error with the menuconfig command

```
C:\esp\iothub_client_sample_mqtt>idf.py menuconfig
Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from C:\esp_idf\requirements.txt are satisfied.
Running cmake in directory C:\esp\iothub_client_sample_mqtt\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 --warn-uninitialized C:\esp\iothub_client_sample_mqtt"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.22.0.windows.1")
-- Building empty aws_iot component due to configuration
CMake Error at C:/esp_idf/components/azure-iot-sdk-c/CMakeLists.txt:5 (project):
  project command is not scriptable
Call Stack (most recent call first):
  C:/esp_idf/tools/cmake/scripts/expand_requirements.cmake:176 (include)
  C:/esp_idf/tools/cmake/scripts/expand_requirements.cmake:183 (expand_component_requirements)
  C:/esp_idf/tools/cmake/scripts/expand_requirements.cmake:286 (expand_component_requirements)

CMake Error at C:/esp_idf/tools/cmake/project.cmake:77 (include):
  include could not find load file:

    C:/esp/iothub_client_sample_mqtt/build/component_depends.cmake
Call Stack (most recent call first):
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1
```

shahpiyushv, 2019-07-18:
@JAHTKELD , as of now, the esp-azure repo does not support Cmake. I will check on adding support for this. For the time being, can you use the release/v3.3 or release/v3.2 branche of esp-idf instead?

JAHTKELD, 2019-07-19:
Hello:
It works!!!. Thank you very much.
 I have used the esp-idf realease/v3.2.2 as explained in the 
**Getting Started Guide for the stable ESP-IDF version** using MSYS2 for Windows 
and I have copied _azure-esp_ repor in the  _esp-idf/components/esp-azure_ folder and I have compiled we make  the iothub_client_sample_mqtt example and it works.

Thanks.

closed 2019-07-22 by JAHTKELD

sagar448, 2019-07-26:
@JAHTKELD Can you please help me? How did you make this work?

kannannair084, 2020-06-17:
@JAHTKELD could you guide on how to add esp_azure component to esp_idf 4.0.1

shahpiyushv, 2020-06-17:
@kannannair084 , the esp-azure works directly with idf 4.0.1 without any changes required. Are yo u facing any issues even after following the READMEs?

kannannair084, 2020-06-17:
hi Piyush, yes indeed i am facing some issues. i got the esp_azure downloaded and moved to component folder of esp_idf 4 ( realease version) and it shows this:

cmake --build . -- -v
[0/1] C:\OLF\Development\Firmware\ESP-IDF\.espressif\tools\cmake\3.13.4\bin\cmake.exe -SC:\OLF\Development\Firmware\OLFirmware-ESP32-LTE-DLC -BC:\OLF\Development\Firmware\OLFirmware-ESP32-LTE-DLC\build
-- Component directory C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4/components/esp-azure does not contain a CMakeLists.txt file. No component will be added
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- ccache will be used for faster recompilation

kannannair084, 2020-06-17:
downloaded it again, and added to components and it still fails, but with a different error.
[683/1106] cmd.exe /C "cd /D C:\OLF\Development\Firmware\OLFirmware-ESP32-LTE-DLC\build\bootloader && C:\OLF\Development\Firmware\ESP-IDF\.espressif\tools\cmake\3.13.4\bin\cmake.exe -DSDKCONFIG=C:/OLF/Development/Firmware/OLFirmware-ESP32-LTE-DLC/sdkconfig -DIDF_PATH=C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4 -DIDF_TARGET=esp32 -DPYTHON=python -DPYTHON_DEPS_CHECKED=1 -DEXTRA_COMPONENT_DIRS=C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4/components/bootloader -DLEGACY_INCLUDE_COMMON_HEADERS=y --warn-uninitialized -GNinja C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4/components/bootloader/subproject && C:\OLF\Development\Firmware\ESP-IDF\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E touch C:/OLF/Development/Firmware/OLFirmware-ESP32-LTE-DLC/build/bootloader-prefix/src/bootloader-stamp/bootloader-configure"
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-configure 
cmd.exe /C "cd /D C:\OLF\Development\Firmware\OLFirmware-ESP32-LTE-DLC\build\bootloader && C:\OLF\Development\Firmware\ESP-IDF\.espressif\tools\cmake\3.13.4\bin\cmake.exe -DSDKCONFIG=C:/OLF/Development/Firmware/OLFirmware-ESP32-LTE-DLC/sdkconfig -DIDF_PATH=C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4 -DIDF_TARGET=esp32 -DPYTHON=python -DPYTHON_DEPS_CHECKED=1 -DEXTRA_COMPONENT_DIRS=C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4/components/bootloader -DLEGACY_INCLUDE_COMMON_HEADERS=y --warn-uninitialized -GNinja C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4/components/bootloader/subproject && C:\OLF\Development\Firmware\ESP-IDF\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E touch C:/OLF/Development/Firmware/OLFirmware-ESP32-LTE-DLC/build/bootloader-prefix/src/bootloader-stamp/bootloader-configure"
Warn about uninitialized values.
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- Project version: v4.0.1-dirty
-- Building ESP-IDF components for target esp32
CMake Error at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4/tools/cmake/component.cmake:222 (message):
  CMake Error at
  C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4/components/azure-iot-sdk-c/CMakeLists.txt:5
  (project):

    project command is not scriptable

  Call Stack (most recent call first):

    C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4/tools/cmake/scripts/component_get_requirements.cmake:60 (include)
    C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4/tools/cmake/scripts/component_get_requirements.cmake:76 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4/tools/cmake/build.cmake:404 (__component_get_requirements)
  C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:27 (project)

-- Configuring incomplete, errors occurred!
See also "C:/OLF/Development/Firmware/OLFirmware-ESP32-LTE-DLC/build/bootloader/CMakeFiles/CMakeOutput.log".
[CMakeOutput.log](https://github.com/espressif/esp-azure/files/4795574/CMakeOutput.log)
[CMakeOutput.log](https://github.com/espressif/esp-azure/files/4795575/CMakeOutput.log)

shahpiyushv, 2020-06-18:
@kannannair084 instead of copying the esp-azure to esp-idf components, have you tried compiling the examples directly from the esp-azure repo? Basically, set IDF_PATH to point to your idf 4.0 and then use `idf.py build` in some example folder (say esp-azure/examples/iothub_client_sample_mqtt).

syamkrishnan-ESL, 2020-06-29:
[error.txt](https://github.com/espressif/esp-azure/files/4844987/error.txt)

syamkrishnan-ESL, 2020-06-29:
I have clonned esp-azure to esp-idf folder and tried to do 
[error.txt](https://github.com/espressif/esp-azure/files/4845013/error.txt)
[error.txt](https://github.com/espressif/esp-azure/files/4845019/error.txt)

syamkrishnan-ESL, 2020-06-29:
[error.txt](https://github.com/espressif/esp-azure/files/4845025/error.txt)

syamkrishnan-ESL, 2020-06-29:
@shahpiyushv and @kannannair084  I have tried to clone esp-azure to esp-idf  and build as shown in log file attached.And getting errors as explained in above discussions.Any suggestions please

shahpiyushv, 2020-06-29:
@syamkrishnan-ESL , are you referring to the logs already posted by others, or your own logs that you may have tried to upload but are missing from your comment?

shahpiyushv, 2020-06-29:
@syamkrishnan-ESL , can you check [this post](https://www.esp32.com/viewtopic.php?f=13&t=16179#p61636) on the forums and see if your issue gets resolved?

syamkrishnan-ESL, 2020-06-29:
attaching logs

[error.txt](https://github.com/espressif/esp-azure/files/4846325/error.txt)

JAHTKELD, 2020-07-06:
**syamkrishnan-ESL** 
Hello: 
In this moment  I have change to use ESP-IDF with  Visual Code Studio and platformio in Windows10
I use the Kentaro example https://github.com/kmwebnet/ECC608-AzureDPS-Provision-idf4 for connect to Azure 

For setup Visual Code Studio and platformio you can use 
https://www.instructables.com/id/Develop-ESP32-With-PlatformIO-IDE/

zafarpatel7, 2020-07-24:
during Clone this esp-azure repository I am facing below issues - 
[image]