espressif/esp-azure #48: Unable to add include iothub_service_client_auth.h to an existing project

lph65724, 2019-07-26:
Hi there,
I am modifying the client sample project.  When I add #include "iothub_service_client_auth.h" to the sample C file, the compiler shows "No such file or directory" unless I have an absolute path.  However, even with absolute path, the compiler still shows undefined reference to 'IoTHubServiceClientAuth_CreateFromConnetionString'.  Where else in the project I need to modify in order to make service_client APIs to work?

thank you

shahpiyushv, 2019-07-30:
@lph65724 it is hard to tell what the exact issue could be. But, can you try with the release/v3.2 or release/v3.3 branches of ESP IDF?

lph65724, 2019-08-08:
Hi @shahpiyushv, 
with v3.2.2, I have run into other compile errors.  I now have 
`*** No rule to make target 'C:/msys32/home/User/esp/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc/uECC.c', needed by 'micro-ecc/uECC.o'.  Stop.`
`make[1]: *** [/home/User/esp/esp-idf/make/project.mk:530: component-micro-ecc-build] Error 2`
`make[1]: *** Waiting for unfinished jobs....`
`make: *** [/home/User/esp/esp-idf/components/bootloader/Makefile.projbuild:41: /home/User/esp/esp-azure/examples/iothub_client_sample_mqtt/build/bootloader/bootloader.bin] Error 2`

Thank you

shahpiyushv, 2019-08-08:
@lph65724 , have you executed `git submodule update --init --recursive` once after switching to v3.2.2?

lph65724, 2019-08-08:
@shahpiyushv, yes, I have.

chinmaychhajed, 2019-08-08:
@lph65724 I regenerated the same issue, following steps helped me to solve it:

1. update toolchain for v3.3
2. `git submodule update --init --recursive` in `$IDF_PAH`
3. navigate to $IDF_PATH/tools/kconfig and run `rm *.d`
4. run `make -j8 clean`
5. then `make -j8 flash monitor`
`$IDF_PATH` is path to my esp-idf directory

these steps should be sufficient

lph65724, 2019-08-08:
@chinmaychhajed after updating to toolchain v3.3 and followed the steps you provided, it worked.  Thank you

closed 2019-08-08 by lph65724