espressif/esp-azure #49: iothub_client.h no such file or directory

sagar448, 2019-07-26:
I cloned the repo with the recursive command into the components folder of the esp-idf. Then i used the iothub_client_sample_mqtt and copied it to my desktop. I ran idf.py menuconfig and it worked fine. But when I ran idf.py build (in the ESP-IDF command prompt) I got error iothub_client.h no such file or directory! Please help, been stuck on this problem for a week!

sagar448, 2019-07-26:
UPDATE

So managed to fix that error. So decided to try a different sample file from the azure iot sdk c. I opened up the prov_dev_client_sample from the provisioning client in the SDK and now I am getting this error. 

[image]

I've tried reconfiguring and rebuilding. Have not been able to solve it yet. Please help!

sagar448, 2019-07-26:
@mahavirj @jitin17 Seeing as you guys are the main contributors, I would really appreciate it if I could help as soon as possible as I have a deadline approaching quite quickly and I've been trying to get azure to work with the esp32 for over 2 weeks! Thank you

shahpiyushv, 2019-07-27:
@sagar448 what version of ESP IDF are you using? Try using release/v3.3 or release/v3.2 branches.

sagar448, 2019-07-27:
Hello @shahpiyushv 

Thanks a lot, I changed the version and it worked. But now there is another problem. Please see the attachment. 

[image]

[image]

Why is it aborting? 

Thanks for your help!

sagar448, 2019-07-27:
The above is after I ran the flash command, uploaded the code.

shahpiyushv, 2019-07-30:
@sagar448 , what example are you trying out? Do you have any modifications in that?

sagar448, 2019-08-01:
Hello @shahpiyushv 

I am trying out the MQTT example in the ESP azure. I've managed to solve the above problem through our email chain. But now i've ran into something else. I have attached it, please check!

[image]

It would be great if you could really help me get to the bottom of this. I have been stuck with the azure sdk for over 3 weeks now, there is hardly any support and the documentation isn't that great. So I would really appreciate it if you could help me through this~!

sagar448, 2019-08-01:
I've made sure my connection string is right, same with the shared access key. Before the SNTP was setting wrong time, I even corrected the time zone. I have no idea what the problem could be. And the fact that the example doesn't work out of the box is very frustrating. @shahpiyushv

shahpiyushv, 2019-08-02:
@sagar448 , it seems that this particular error code (IOTHUB_CLIENT_CONNECTION_DEVICE_DISABLED) means : "Device could not be authenticated because it is disabled by the user on the Azure IoT Hub (see field State in Device Explorer)". Can you check this on the Azure portal once?

sagar448, 2019-08-02:
Hello @shahpiyushv 

Thank you for getting back to me. I did see that before, that device disabled and I did check the Azure Hub, the device is enabled yet I get that error. Please have a look below. 
[image]

Any other suggestion? It would be great if i could maybe send you my code and you can check it out?

Thanks, please let me know!

sagar448, 2019-08-04:
Hey guys @shahpiyushv @mahavirj ,

Any updates please?

Thank you!

shahpiyushv, 2019-08-04:
@sagar448 , it is hard to tell if the problem is on the device side or the Azure side. Since this is just a secure MQTT connection, can you try some other means to connect to the Azure IoT Hub like mosquitto (https://mosquitto.org/man/mosquitto_pub-1.html), MQTT Fx (https://mqttfx.jensd.de/) or python (https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-mqtt-support#using-the-mqtt-protocol-directly-as-a-module)? That will help isolating the problem better.

sagar448, 2019-08-11:
You can close this because I switched to AWS

closed 2019-08-11 by sagar448