espressif/esp-azure #50: OTA update with Azure

elieDaan, 2019-08-01:
Hello, 

I'm working on a secure solution to do OTA update. I've already succeed to update my code with OTA API provided by ESP-IDF. 

Do you think it's possible to do it with Azure C SDK like this exemple with [Azure Node JS](https://docs.microsoft.com/en-us/azure/iot-hub/tutorial-firmware-update). If yes, will it be more secured than esp-idf ?

shahpiyushv, 2019-08-02:
@elieDaan I think there could be some confusion in this. The ESP IDF OTA APIs primarily focus on fetching the image data over standard HTTPS, writing it to the device flash, verifying the image and changing the boot partition. Azure, on the other hand would focus on the server side, primarily hosting the image and then somehow providing  the image/url to the device by using their C agent. So, an actual OTA implementation would be a combination of ESP IDF OTA + Azure C SDK.

shahpiyushv, 2019-10-18:
@elieDaan , I checked the node.js example link above and it seems that it is just using the Device Twin to get FW upgrade info (url, version, etc.) and also report the upgrade status. You can try out the device twin example in esp-azure for that. The actual OTA will still require some file hosting service, and you can fetch image from that using esp_https_ota. Since this part is not really relevant in this project, I will close this issue.

closed 2019-10-18 by shahpiyushv