espressif/esp-azure #51: After updating: esp-tls: mbedtls_ssl_handshake returned -0x4c

cheezum, 2019-08-07:
Maybe this might be the wrong place, but:
after updating the esp-idf and the esp-azure my working code is not able to connect to azure anymore and returns a -0x4c code.

> esp-tls: mbedtls_ssl_handshake returned -0x4c

What I suspect is that the certificates are not correct. I tried to replace the certs.c in the port/src directory with the one from azure-iot-sdk-c/certs version but without any luck.

Is this an Azure-SDK thing or do I need to look at the esp-tls module?

nrmka, 2019-08-07:
Hi @cheezum , I have the same issue. Which branch in esp-idf do you stay at? At the latest master I get this error. Otherwise, release/v3.3 or v3.2 works fine for me. I tried to investigate the problem and stopped somewhere at the socket level:

getsockopt(fd, SOL_SOCKET, SO_ERROR, &sock_errno, &optlen);

returns zero (the newest version) instead EAGAIN (release version), so the program stops to wait/try-again-to-read response from the server and fails handshake.

cheezum, 2019-08-07:
I'm using the master (v4.0-dev-1434).
I'll try to revert and let you know what happened.

chinmaychhajed, 2019-08-08:
try compiling on v3.3 only as support for any release of v4.0 is not yet added. For reference you can check !48. Generally these steps solves many issues

chinmaychhajed, 2019-08-08:
try compiling on v3.3 only as support for any release of v4.0 is not yet added. For reference you can check [this](https://github.com/espressif/esp-azure/issues/48#issuecomment-519449579). Generally these steps solves many issues

cheezum, 2019-08-08:
Does it mean I have to revert back to 'make'  instead of CMake?
It was pretty much a pain to switch to CMake and I don't want to go back to Make.
Now when I try to build (with CMake) on v3.3 with the latest esp-azure it gives me a lot of missing header files.

chinmaychhajed, 2019-08-08:
did you try `git submodule update --init --recursive`?

chinmaychhajed, 2019-08-08:
Support for `CMake` is not added yet. It will be added in future versions. So currently you will have to use `make` only.

cheezum, 2019-08-08:
Yes I did.
My folder structure is like this:
~/esp
~/esp/Interface2/
~/esp/esp-idf/
~/esp/esp-idf/components/
~/esp/esp-idf/components/esp-azure

To fetch the v3.3 I did:
`git clone --single-branch --branch release/v3.3 https://github.com/espressif/esp-idf.git --recursive`
After that I cloned the esp-azure into components/
~/esp/esp-idf/components > git clone --recursive https://github.com/espressif/esp-azure

After cloning, I need to copy the CMakelists.txt file from esp-azure/port to esp-azure/ and modify it to:
> 
> set (AZURE_IOT_SDK "${CMAKE_CURRENT_LIST_DIR}/azure-iot-sdk-c")
> 
> set (COMPONENT_ADD_INCLUDEDIRS
>     "port/inc"
>     "${AZURE_IOT_SDK}/certs"
>         "${AZURE_IOT_SDK}/c-utility/inc"
>         "${AZURE_IOT_SDK}/c-utility/inc/azure_c_shared_utility"
>         "${AZURE_IOT_SDK}/c-utility/pal/inc"
>         "${AZURE_IOT_SDK}/c-utility/pal/freertos"
>         "${AZURE_IOT_SDK}/c-utility/pal/generic"
>         "${AZURE_IOT_SDK}/iothub_client/inc"
>         "${AZURE_IOT_SDK}/serializer/inc"
>         "${AZURE_IOT_SDK}/umqtt/inc"
>         "${AZURE_IOT_SDK}/umqtt/inc/azure_umqtt_c"
>         "${AZURE_IOT_SDK}/deps/parson"
>         "${AZURE_IOT_SDK}/provisioning_client/inc"
>         "${AZURE_IOT_SDK}/provisioning_client/adapters"
>         "${AZURE_IOT_SDK}/provisioning_client/deps/utpm/inc"
>         )
> 
> set (COMPONENT_SRCS
>         "port/src/agenttime_esp.c"
>         "port/src/platform_esp.c"
>         "port/src/tlsio_esp_tls.c"
>         "${AZURE_IOT_SDK}/certs/certs.c"
>         "${AZURE_IOT_SDK}/c-utility/pal/freertos/lock.c"
>         "${AZURE_IOT_SDK}/c-utility/pal/dns_async.c"
>

Here, I removed the ../ from the CMAKE_CURRENT_LIST line and place port/ in front of the src and inc folder.

After this I try to compile but then the compiler is unable to find nvs_flash.h, nvs.h and iothub_client.h.
The files are all referenced by custom code (~/esp/Interface2/main/helpers/)

The examples from both ESP-IDF and ESP-AZURE do compile.

closed 2019-08-21 by igrr