espressif/esp-azure #52: AMQP on ESP-Azure-IDF

SeriousD, 2019-08-13:
Hello,
I would like to use the AMQP message protocol, for connecting multiple devices via CAN-Bus through a single ESP32 as a gateway.
I included 
`#include "iothubtransportamqp.h"`
and set 
`IoTHubClient_LL_CreateFromConnectionString(connectionString, AMQP_Protocol)`
for testing purposes but the compiler throws:
>  undefined reference to `AMQP_Protocol'

Am I doing something wrong and if yes: whats the proper way to use AMQP?

chinmaychhajed, 2019-08-13:
Hello, thanks for reporting this issue. Current version of esp-azure does not support use of AMQP protocol, although it is supported by Microsoft Azure. Support of AMQP will be added in further versions if we receive multiple requests for it. Instead of AMQP, you can use either HTTP or MQTT protocol.

shahpiyushv, 2019-10-18:
@SeriousD , as @chinmaychhajed mentioned , we may take up this request later. However, can you tell me what exactly is the use case for using AMQP, as against MQTT?

SeriousD, 2019-10-20:
Hello,
yes, I want to use the ESP as kind of field gateway, that translates from a CAN network to the IoT hub. For this I want to establish one connection for each CAN Device. If I understand correctly this is connection multiplexing. According to [this](https://docs.microsoft.com/de-de/azure/iot-hub/iot-hub-devguide-protocols) Azure documentation, AMQP is the best protocol for this use case.

Correct me if I'm wrong or you have a better idea.

arnowi, 2020-03-23:
It would be very much appreciated if amqp over websockets is supported. I'm currently trying to get it to work for me but it's a slow process.

Darkenkade42, 2020-04-20:
What is needed to get AMQP working? I'd like some high level guidance if possible.

shahpiyushv, 2020-04-22:
Azure supports too many combinations, and I haven't had a chance to check and add support for all, since there are multiple dependencies. If anyone has already worked on it and can provide some patches, I can pull them in.