espressif/esp-azure #53: Support for current IDF

jubueche, 2019-08-26:
Hi,

this library is really cool, but ESP is moving forward with their IDF version so it would be nice if we could use a newer IDF version. I tried it but it gave me a lot of compiling errors.

chinmaychhajed, 2019-08-27:
We will be adding support of newer IDF versions, but currently all the testing has been done using [IDF release/v3.3](https://github.com/espressif/esp-idf/tree/release/v3.3). Please use this version and let us know if you still face any problems.

NielsVeul, 2019-08-28:
Hi, do you have any idea on when support for IDF v4.0 will be added? And will the next release work with a newer version of the azure-iot-sdk-c?

chinmaychhajed, 2019-08-28:
It is in plan but not yet a priority item, as IDF 4.0 will need some time to stabilise first. Whenever that happens, we will also ensure that the latest  azure-iot-sdk-c is used

fromeijn, 2019-09-03:
would like to see this happen any time soon :wink:

shahpiyushv, 2019-10-15:
@jubueche @fromeijn @NielsVeul , the esp_azure has been updated. It can now compile with IDF tag: v4.0-beta1

jubueche, 2019-10-15:
Closing this. Any issues appearing would be separate ones.

closed 2019-10-15 by jubueche