espressif/esp-azure #54: Use with PlatformIO

vipex, 2019-09-06:
Hello,

I'm trying to use this in a project generated with PlatformIO ( espressif32 / espidf )
tried with various combination of configuration ( clone this in /lib, configuring .ini, placing in /src/components ) but in the end I'm stuck on an Azure SDK error:

```
*** Multiple ways to build the same target were specified for: /home/user/dev/TestAzure/.pio/build/esp32dev/src/components/esp-azure/azure-iot-sdk-c/iothub_client/tests/iothubtransporthttp_ut/iothubtransporthttp_ut.o  (from ['/home/user/dev/TestAzure/src/components/esp-azure/azure-iot-sdk-c/iothub_client/tests/iothubtransporthttp_ut/iothubtransporthttp_ut.c'] and from ['/home/user/dev/TestAzure/src/components/esp-azure/azure-iot-sdk-c/iothub_client/tests/iothubtransporthttp_ut/iothubtransporthttp_ut.cpp'])
File "/home/user/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 308, in BuildSources
```

Someone can help me?
( hope it's not a completely dumb question, I'm really new in this so, please, be merciful 😅)

thanks in advance.

chinmaychhajed, 2019-09-06:
Hello,

It looks like a build failure problem. Try following [these](https://github.com/espressif/esp-azure/issues/46#issuecomment-526908813) general steps/precautions in case of build failure. After following these steps you should be able to build your project properly.

d-e-m, 2019-10-04:
I am have the same question for using Platform.io for importing, building and flashing these examples. Do anyone have some documentation around that?

andriyadi, 2020-04-15:
My team can make it works with latest esp-idf 4 and PlatformIO. It involves the changes in CMakeList. 
Take a look a complete example, with the changed `esp-azure` is bundled in it: [https://github.com/dycodex/esp32-azure-iot-training/tree/master/day3/finished/esp32_azure_iot_hub](https://github.com/dycodex/esp32-azure-iot-training/tree/master/day3/finished/esp32_azure_iot_hub)