espressif/esp-azure #55: Fail to connect to Azure IoT hub: esp-tls: couldn't get hostname for :chaze-iot-hub.azure-devices.net:

jubueche, 2019-09-07:
Hi,

I am executing code that used to work in February, but not anymore. I am now getting the following error:

```D (12903) esp-tls: host:chaze-iot-hub.azure-devices.net: strlen 31
I (18813) wifi: bcn_timout,ap_probe_send_start
E (19143) esp-tls: couldn't get hostname for :chaze-iot-hub.azure-devices.net:```

My trace looks like this:

```
V (9853) event: exit default callback
D (10643) tcpip_adapter: if0 dhcpc cb
D (10643) tcpip_adapter: if0 ip changed=1
D (10643) event: SYSTEM_EVENT_STA_GOT_IP, ip:192.168.178.87, mask:255.255.255.0, gw:192.168.178.1
V (10643) event: enter default callback
I (10653) event: sta ip: 192.168.178.87, mask: 255.255.255.0, gw: 192.168.178.1
V (10653) event: exit default callback
I (10663) Chaze-WIFI-Synch: SYSTEM_EVENT_STA_GOT_IP
I (10663) Chaze-WIFI-Synch: got ip:192.168.178.87

I (10673) Chaze-WIFI-Synch: Connected to WiFi, syncing data now...
I (10683) Chaze-WIFI-Synch: Azure task got the lock...
Initializing SNTP
ESP platform sntp inited!
Time is not set yet. Connecting to WiFi and getting time over NTP. timeinfo.tm_year:70
Waiting for system time to be set... tm_year:0[times:1]
I (12703) platform: The current date/time is: Sat Sep  7 17:28:10 2019
I (12863) Chaze-WIFI-Synch: Set time to 07/09/2019
I (12863) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (12863) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (12873) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (12903) esp-tls: host:chaze-iot-hub.azure-devices.net: strlen 31
I (18813) wifi: bcn_timout,ap_probe_send_start
E (19143) esp-tls: couldn't get hostname for :chaze-iot-hub.azure-devices.net:
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
I (21313) wifi: ap_probe_send over, resett wifi status to disassoc
I (21313) wifi: state: run -> init (c800)
I (21313) wifi: pm stop, total sleep time: 0 us / 11543320 us

I (21313) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
D (21323) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:EatOrBeEaten, ssid_len:12, bssid:e8:df:70:85:8f:76, reason:200,wifi reason: beacon timeout
V (21333) event: enter default callback
V (21343) tcpip_adapter: check: local, if=0 fn=0x4011d554
0x4011d554: tcpip_adapter_down_api at /home/julian/esp/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:339

D (21343) tcpip_adapter: if0 start ip lost tmr: enter
Info: Waiting for TLS connection
D (21353) tcpip_adapter: if0 start ip lost tmr: interval=120
V (21363) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (21363) tcpip_adapter: check: remote, if=0 fn=0x4011d554
0x4011d554: tcpip_adapter_down_api at /home/julian/esp/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:339

V (21373) esp_adapter: thread sem get: sem=0x3ffe1c5c
V (21373) event: exit default callback
I (21383) Chaze-WIFI-Synch: SYSTEM_EVENT_STA_DISCONNECTED
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
Info: Waiting for TLS connection
```

This is my code for uploading a file to Azure:

```
/**
 * @brief Initializes the IoTHub client, sets options, sets filename, uploads data to blob.
 * @param pvParameter
 */
void synch_with_azure(void *pvParameter)
{

	for(;;){
		if( xSemaphoreTake( xSemaphore, LONG_TIME ) == pdTRUE )
		{
			ESP_LOGI(TAG_WiFi, "Azure task got the lock...");
			break;
		}
		vTaskDelay(100 / portTICK_PERIOD_MS);
		ESP_LOGI(TAG_WiFi, "Azure task waiting for lock...");
	}

	(void)IoTHub_Init();
	device_ll_handle = IoTHubDeviceClient_LL_CreateFromConnectionString(ft.get_azure_connection_string(), HTTP_Protocol);

	if (device_ll_handle == NULL)
	{
		ESP_LOGE(TAG_WiFi, "Failure creating Iothub device. Check conn string.");
	}
	else
	{

		IoTHubDeviceClient_LL_SetOption(device_ll_handle, OPTION_TRUSTED_CERT, certificates);
		if(config.STATE == ADVERTISING || config.STATE == CONNECTED)
		{

			time_t now;
			time_t tmp;
			now = get_time(&tmp);

			char strftime_buf[64];
			struct tm timeinfo;

			localtime_r(&now, &timeinfo);
			strftime(strftime_buf, sizeof(strftime_buf), "%c", &timeinfo);

			if(rtc.begin()) {
				rtc.set24Hour();
				rtc.setTime(timeinfo.tm_sec, timeinfo.tm_min, timeinfo.tm_hour, timeinfo.tm_wday, timeinfo.tm_mday, timeinfo.tm_mon +1, timeinfo.tm_year +1900);	
				ESP_LOGI(TAG_WiFi, "Set time to %s", rtc.stringDate());
			} else {
				ESP_LOGE(TAG_WiFi, "Error setting time.");
			}

			for(int i=0;i<strlen(strftime_buf);i++){
				if(strftime_buf[i] == ' '){
					strftime_buf[i] = '_';
				}
			}
			char file_name[128]; //64+32+puffer
			strcpy(file_name, ft.get_device_id());
			strcat(file_name, "_");
			strcat(file_name, strftime_buf);
			strcat(file_name, ".txt");

			ft.start_reading_data();

			if (IoTHubDeviceClient_LL_UploadMultipleBlocksToBlob(device_ll_handle, file_name, getDataCallback, NULL) != IOTHUB_CLIENT_OK)
			{
				ESP_LOGE(TAG_WiFi, "Failed to upload");
			}
			else
			{
				ESP_LOGI(TAG_WiFi, "Successful upload.");
			}
		}

		IoTHubDeviceClient_LL_Destroy(device_ll_handle);
	}
	IoTHub_Deinit();
	if(xSemaphoreGive(xSemaphore) != pdTRUE)
	{
		ESP_LOGE(TAG_WiFi, "Failed to release the semaphore.");
	}
	ESP_LOGI(TAG_WiFi, "Delete this task.");
	vTaskDelete(NULL);
}
```

In my main folder I have a folder named server_certs with the msft_ssl_certificate.pem .
After some time the connection is interrupted, which I also don't understand.

If it helps, I could attach my WiFi event handler.

shahpiyushv, 2019-10-15:
@jubueche can you double check that your IOT Hub (chaze-iot-hub.azure-devices.net) is indeed reachable from a Host machine in the same network? You can use the monitor events command from https://github.com/espressif/esp-azure#monitoring-results for that.

jubueche, 2019-10-16:
Unfortunately this functionality does not exist anymore. I downloaded the Azure CLI and the command cannot be found. Any suggestions?

shahpiyushv, 2019-10-16:
@jubueche , have you added the Azure IoT CLI extension using `az extension add --name azure-cli-iot-ext` as mentioned here: https://github.com/espressif/esp-azure#setting-up-azure-cli?

jubueche, 2019-10-20:
Turns out that it was a memory issue.

closed 2019-10-20 by jubueche