espressif/esp-azure #56: prov_dev_client_ll_sample not working with CMake

fromeijn, 2019-09-12:
tried this example, this is working with Make but not with CMake. I would like to use CMake as compile times are shorter. Get the following output:
```
[47/54] Linking C static library esp-idf/main/libmain.a
[48/54] Generating idf_component_main.sections_info
[49/54] Building C object esp-idf/bootloader_support/CMakeFiles/idf_component_bootloader_support.dir/src/bootloader_init.c.obj
[50/54] Linking C static library esp-idf/bootloader_support/libbootloader_support.a
[51/54] Generating idf_component_bootloader_support.sections_info
[52/54] Building C object CMakeFiles/bootloader.elf.dir/dummy_main_src.c.obj
[53/54] Linking C executable bootloader.elf
[54/54] Generating ../bootloader.bin
esptool.py v2.8-dev
[911/912] Linking CXX executable prov_dev_client_ll_sample.elf
FAILED: prov_dev_client_ll_sample.elf 
: && /home/floris/.espressif/tools/xtensa-esp32-elf/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls  -nostdlib CMakeFiles/prov_dev_client_ll_sample.elf.dir/dummy_main_src.c.obj  -o prov_dev_client_ll_sample.elf  -Wl,--cref -Wl,--Map=prov_dev_client_ll_sample.map -Wl,--gc-sections -Wl,--start-group esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/xtensa-debug-module/libxtensa-debug-module.a esp-idf/app_trace/libapp_trace.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/esp_event/libesp_event.a esp-idf/ethernet/libethernet.a esp-idf/mbedtls/libmbedtls.a esp-idf/micro-ecc/libmicro-ecc.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/pthread/libpthread.a esp-idf/smartconfig_ack/libsmartconfig_ack.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/cxx/libcxx.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/port/libport.a esp-idf/asio/libasio.a esp-idf/jsmn/libjsmn.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/openssl/libopenssl.a esp-idf/esp_https_server/libesp_https_server.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/fatfs/libfatfs.a esp-idf/freemodbus/libfreemodbus.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mdns/libmdns.a esp-idf/mqtt/libmqtt.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/main/libmain.a -lgcov esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a -Wl,--undefined=uxTopUsedPriority esp-idf/newlib/libnewlib.a -L /home/floris/Documents/git/esp-idf-v3.3/components/newlib/lib -lc -lm esp-idf/esp32/libesp32.a -L /home/floris/Documents/git/esp-idf-v3.3/components/esp32/lib -lcoexist -lcore -lespnow -lmesh -lnet80211 -lphy -lpp -lrtc -lsmartconfig -lwpa2 -lwpa -lwps -L /home/floris/Documents/git/esp-azure/examples/prov_dev_client_ll_sample/build/esp-idf/esp32 -T esp32_out.ld -T /home/floris/Documents/git/esp-azure/examples/prov_dev_client_ll_sample/build/esp-idf/esp32/esp32.project.ld -L /home/floris/Documents/git/esp-idf-v3.3/components/esp32/ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld /home/floris/Documents/git/esp-idf-v3.3/components/esp32/libhal.a -lgcc -u call_user_start_cpu0 -u ld_include_panic_highint_hdl esp-idf/cxx/libcxx.a -lstdc++ -u __cxa_guard_dummy -u __cxx_fatal_exception -u esp_app_desc && :
esp-idf/main/libmain.a(custom_hsm_x509.c.obj):(.literal.custom_hsm_create+0x4): undefined reference to `_binary_leaf_certificate_pem_start'
esp-idf/main/libmain.a(custom_hsm_x509.c.obj):(.literal.custom_hsm_create+0x8): undefined reference to `_binary_leaf_private_key_pem_start'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

chinmaychhajed, 2019-09-23:
Hello,

The certificates are not embedded in the device using CMake. After applying [this](https://github.com/espressif/esp-azure/files/3641812/changes.patch.txt)
 patch, the example should work.

fromeijn, 2019-09-24:
Thanks will try that!

closed 2019-09-24 by fromeijn