espressif/esp-azure #58: How to use ATECC608 with Azure?

SrLeon, 2019-09-17:
I doing a IoT project where I have to connect ESP32 to Azure, but is important to have a secure connection and a secure device. I am trying to use an external ATECC608 crypto memory. But is being a nightmare trying to make it work.

I am using the https://github.com/MicrochipTech/cryptoauthlib library from microchip and it works perfectly as same as examples to connect ESP32 to Azure.

How I can make it work all together?

fromeijn, 2019-10-30:
would like to see an example for this

wouterdevinck, 2019-12-03:
This is indeed not so straight forward, but I have gotten it to work.

It's not a full write-up yet, but the pointers in here should help:
https://github.com/espressif/esp-idf/issues/4432

SalmanMalik7183, 2023-06-22:
Hi @SrLeon, 
its been a while since this issue was clsed but were you able to connect to azure using the certificates from secure element. I am working on a similar project and not having any luck exactly. It you could help out that'd be great. lemme know when you see this.
thanks