espressif/esp-azure #59: #unable to connect esp32-azure using c sdk

enerdata, 2019-10-01:
Hi there,

iam working on esp32 azure ,unable to connect azure and the error is

 Func:tlsio_esp_tls_destroy Line:152 tlsio_esp_tls_destroy called while not in TLSIO_STATE_CLOSED.

shahpiyushv, 2019-10-15:
@enerdata , which example are you using? We have updated some documentation right now. Can you try as per that documentation?

enerdata, 2019-10-17:
yeah !!

i will try now 
Thank you @shahpiyushv

enerdata, 2019-11-05:
Hi @shahpiyushv 
i cloned the updated git , but still iam facing the same issue can you please guide to resolve it ?

shahpiyushv, 2019-11-05:
@enerdata , is this related to https://github.com/espressif/esp-azure/issues/66 ? which example are you using?

enerdata, 2019-11-06:
Hi @shahpiyushv  ,
Greetings of the day!!!

i'am using this example iothub_client_sample_mqtt..

enerdata, 2019-11-06:
[image]
please check that image ....... iam getting the above error..

shahpiyushv, 2019-11-06:
@enerdata , does your issue seem somewhat similar to this: https://github.com/espressif/esp-idf/issues/3192? The fix for this is already available on the ESP IDF master branch now.

enerdata, 2019-11-07:
Hi @shahpiyushv 

error resolved ...

Thank you!!!

closed 2019-11-07 by enerdata