espressif/esp-azure #60: Sample code with devicetwin various issues

liudr, 2019-10-03:
Hi,
I am very new to this but I'm learning. I have azure account, created a device. Have host name, device id and sharedaccesskey. 

Everything looks good on azure. I can see under devicetwin that the "car" shows up with all its properties, make, oil change, etc.

Now the issues:
I am trying to update devicetwin say the maxspeed. I went in and change maxspeed to 70, save it. A second after saving, it reverts to the 100, which is the value initially sent by esp.

On esp side, mqtt seems to have trouble connecting:
Error: Time:Thu Oct  3 04:58:37 2019 File:D:/Downloads/msys32_iot/home/user/esp/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c Func:mqtt_error_callback Line:1864 Mqtt Ping Response was not encountered.  Reconnecting device...
This pops up once in a while, meaning esp keeps trying but mqtt just won't work.

I also tried direct method. The sample code seems to have getCarVIN method to return a VIN and returns { } for all other methods. I tried this method several times and got null. Coincidentally, the esp side shows:
E (246635) esp-tls: write error :-80:
Info: Error from SSL_write: -80
E (246645) esp-tls: write error :-80:
Info: Error from SSL_write: -80

I've downloaded the latest msys (2018-10) and V3.3 of esp-idt and followed the install instructions. I really appreciate if someone could give me a hand here.

liudr, 2019-10-03:
Which version is esp-idf and esp-mqtt in when this repo was last updated? How do I use those older versions? I could use some help with the git commands. Thanks.

shahpiyushv, 2019-10-15:
@liudr , which value are you changing? desired_maxSpeed or reported_maxSpeed?

shahpiyushv, 2020-06-04:
@liudr , I didn't hear back on this. A bit too late to ask, but is this still valid?