espressif/esp-azure #62: MQTT over websocket not compiling

liudr, 2019-10-03:
If I enable the SAMPLE_MQTT_OVER_WEBSOCKETS option, the device twin sample code fails to compile:

C:/Users/user/Downloads/msys32/home/user/esp/esp-azure/examples/iothub_devicetwin_samples_and_methods/build/main\libmain.a(iothub_client_device_twin_and_methods_sample.o):(.literal.iothub_client_device_twin_and_methods_sample_run+0x4): undefined reference to `MQTT_WebSocket_Protocol'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/Users/user/Downloads/msys32/home/qk9446me/esp/esp-idf/make/project.mk:483: /home/user/esp/esp-azure/examples/iothub_devicetwin_samples_and_methods/build/iothub_devicetwin_samples_and_methods.elf] Error 1

Seems like this protocol is not found. What am I missing? Thanks.

shahpiyushv, 2019-10-18:
@liudr , even I tried giving this a shot, trying to fix some compilation issues, but there seem to be quite some dependencies. We may take this up later.

acheve, 2019-10-22:
@shahpiyushv ,
It's really a problem for us to be stuck to a classical MQTT connection to port 8883.
Some corporate firewalls and home routers could block outbound traffic on "specific" ports.
It would be a true relief to be able to use MQTT over WebSocket (with output port : 443).
Do you have any plan to addd support of MQTT over WebSocket soon?
Thanks!!

daniel-grabner, 2020-01-16:
@shahpiyushv 
For us this is also a huge problem as in many customer installations port 8883 is blocked. In countries such as Switzerland most ISPs have blocked this port by default and charge for opening it. Thus MQTT over websocket is an essential feature for us.

If the problem is not going to be solved in esp-azure port we need to look for alternatives. Talking about 20k of ESP32 per annum. Can you give an estimation of when MQTT over websockets is going to be implemented by Espressif?

liudr, 2020-04-14:
Any update on the possibility of MQTT_OVER_WEBSOCKETS? Thanks.

daniel-grabner, 2020-04-15:
> Any update on the possibility of MQTT_OVER_WEBSOCKETS? Thanks.

+1
Urgently waiting for it too.

Adamantinu, 2020-04-16:
+1
Would be great to have that feature here too, since many firewalls in various locations block the standard mqtt port per default and need to be opened then by a sysadmin if IoT Devices would like to operate there with the ESP azure sdk

MartinTJDK, 2020-07-02:
Fixing issue described in https://github.com/espressif/esp-azure/issues/90 and adding:

const IO_INTERFACE_DESCRIPTION* socketio_get_interface_description(void)
{
    return wsio_get_interface_description();
}

to platform code fixes compilation, and allows MQTT over WebSocket.

Lisa999, 2020-08-14:
> const IO_INTERFACE_DESCRIPTION* socketio_get_interface_description(void)
> {
> return wsio_get_interface_description();
> }

And #include "azure_c_shared_utility/wsio.h" ?

giroudon, 2020-09-11:
MQTT over WebSocket now works on my cloned repo. Please see Pull request #68, which now includes "Upload to Blob" and "MQTT over WebSocket" features.

jspngh, 2021-01-18:
MQTT over websockets works for me when following @MartinTJDK's instructions
and adding some files to `port/CMakeLists.txt`:
```
"${AZURE_IOT_SDK}/iothub_client/src/iothubtransportmqtt_websockets.c"
"${AZURE_IOT_SDK}/c-utility/src/gb_rand.c"
"${AZURE_IOT_SDK}/c-utility/src/utf8_checker.c"
"${AZURE_IOT_SDK}/c-utility/src/uws_client.c"
"${AZURE_IOT_SDK}/c-utility/src/uws_frame_encoder.c"
"${AZURE_IOT_SDK}/c-utility/src/wsio.c"
```
Is there a reason these files are absent and am I supposed to add them somewhere else?