espressif/esp-azure #63: Is this repo supported?

mpinkus, 2019-10-03:
Not a lot of commitment to the platform, can we get some ESP support? Sure would like to buy a lot of your chips....

shahpiyushv, 2019-10-04:
@mpinkus, this repo is indeed supported. There have been some delays due to internal reassigning of resources. But, we will see to it that all the pending issues are looked into and resolved as soon as possible.

mpinkus, 2019-10-04:
Well I need working examples of message send and receive, direct method receive and twin property updates. I'm making a decision a 100k+ unit buy. Can I get a direct contact for my lead developer?

Sent from my Verizon, Samsung Galaxy smartphone

liudr, 2019-10-06:
@shahpiyushv Since you know the inner workings of the project/espressif, could you comment on the following. I appreciate your help!

Our goal is to have the following working:
Based on the device twin sample code:

MQTT over socket option in sample code 
        No other options will compile unless I only use MQTT but I'm worried without sockets there is no response received. I got SSL error and no response from pining info on the device that could send in initial device twin properties.
Device twin update
        Your SDK says sending desired and reported version under development?
Device twin update notification subscription and callback on notification
Method
        There is a getCarVIN method already in the code that I thought should work. I added debug print and this callback never got invoked when I used direct method from azure, and your sdk says it should work.
Device-to-cloud message
Cloud-to-device message
        Doesn't seem to have a handler in sample code
        *IoT Hub supports the option to complete/reject/abandon C2D messages over HTTPS and AMQP only at the moment. So I can't use MQTT for this?

Also some simple wrapper functions to POST to https server and a simple web server for getting SSID/PASS etc.

Your sdk:
https://github.com/Azure/azure-iot-sdk-c/tree/5d9a19a5e5ce6f4974d333bd28f432ee9284f70b

shahpiyushv, 2019-10-17:
@liudr , I tested the devicetwin example myself and I have also updated the documentation. Can you check once again, following the exact documentation at https://github.com/espressif/esp-azure/blob/master/examples/iothub_devicetwin_samples_and_methods/README.md

mpinkus, 2019-10-31:
is there a working version of mqtt anywhere? is anybody supporting any of these sdk's?

don't point me to another web page..respond with a concise example of restarting wifi and having the sntp connection surviving this most basic scenario...

Error: Time:Thu Oct 31 18:07:24 2019 File:C:\Users\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\AzureIoT\src\az_iot\c-utility\pal\src\tlsio_openssl_compact.c Func:tlsio_openssl_destroy Line:213 tlsio_openssl_destroy called while not in TLSIO_STATE_CLOSED.

Info: Initializing SNTP

assertion "Operating mode must not be set while SNTP client is running" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/apps/sntp/sntp.c", line 600, function: sntp_setoperatingmode
abort() was called at PC 0x4012d35f on core 1

mpinkus, 2019-10-31:
i'm trying to restart (close/reset/?anything?) mqtt after a wifi reset

if(WiFi.status() != WL_CONNECTED) 
    { 
      LogInfo("Wifi Disconnected."); 

      ConnectWifi();
      
      LogInfo("Esp32MQTTClient_Reset. Start."); 
      Esp32MQTTClient_Reset();
      LogInfo("Esp32MQTTClient_Reset. Complete");
       
      //ConnectIot();
    }

shahpiyushv, 2019-11-01:
@mpinkus , yes, this project is being maintained actively. However, this is supposed to be used with ESP IDF. The logs I see seem to be for Arduino. Azure for Arduino is maintained separately by Microsoft at https://github.com/VSChina/ESP32_AzureIoT_Arduino. Can you double check if you are indeed using the correct examples and azure agent?

Meanwhile, for SNTP to work properly, please ensure that your code initialises SNTP as below:

```
if (!sntp_enabled()) {
    sntp_setoperatingmode(SNTP_OPMODE_POLL);
    sntp_setservername(0, SNTP_SERVER_NAME);
    sntp_init();
}
```
This will ensure that sntp_setoperatingmode() will not get called while SNTP is already running, thereby preventing that crash.

shahpiyushv, 2020-07-02:
Closing this issue as there isn't anything pending here.

closed 2020-07-02 by shahpiyushv