espressif/esp-azure #64: prov_dev_client_ll_sample - json auth key errors

albahmed23, 2019-10-22:
Hello,

After running the prov_dev_client_ll_sample using x509 cert to authenticate to dps. The following error is puzzling us and we do not know the root cause. Is it possible for someone to help us please?

This is the error we get:
```
<- 14:36:06 PUBLISH | IS_DUP: false | RETAIN: 0 | QOS: DELIVER_AT_LEAST_ONCE = 0x01 | TOPIC_NAME: $dps/registrations/res/401/?$rid=1 | PACKET_ID: 2 | PAYLOAD_LEN: 156
Error: Time:Tue Oct 22 14:36:06 2019 File:*/esp-azure/azure-iot-sdk-c/provisioning_client/src/prov_device_ll_client.c Func:prov_transport_process_json_reply Line:323 failure retrieving json auth key value
Error: Time:Tue Oct 22 14:36:06 2019 File:*/esp-azure/azure-iot-sdk-c/provisioning_client/src/prov_transport_mqtt_common.c Func:prov_transport_common_mqtt_dowork Line:965 Unable to process registration reply.
I (7640) AZURE_LIB: Doing SDK Work
Error: Time:Tue Oct 22 14:36:06 2019 File:*/esp-azure/azure-iot-sdk-c/provisioning_client/src/prov_device_ll_client.c Func:on_transport_registration_data Line:572 Failure retrieving data from the provisioning service
Failure encountered on registration PROV_DEVICE_RESULT_PARSING
E (7680) esp-tls: write error :-80:
Info: Error from SSL_write: -80
-> 14:36:06 DISCONNECT
Error: Time:Tue Oct 22 14:36:06 2019 File:*/esp-azure/port/src/tlsio_esp_tls.c Func:tlsio_esp_tls_destroy Line:152 tlsio_esp_tls_destroy called while not in TLSIO_STATE_CLOSED.
I (7700) AZURE_LIB: Registeration Failed...
```

shahpiyushv, 2019-10-23:
@albahmed23 , have you followed the exact instructions from https://github.com/espressif/esp-azure/tree/master/examples/prov_dev_client_ll_sample? There could be a problem with some configuration value.

albahmed23, 2019-10-23:
The only difference is that I am using group enrollments and connecting to IoT Central. That shouldn’t affect it though? As I know that IoT Central uses DPS to register devices.

albahmed23, 2019-10-23:
Found the issue - it was with the x509 creation step that was wrong. Used openssl to fix the issue. Thanks!

closed 2019-10-23 by albahmed23