espressif/esp-azure #67: Cloning the repo fails with... fatal: bad object 3427761cdc289eccefb88890410f1bf17bf3c4f6

whitsonk, 2019-11-04:
From Powershell...
```
>git clone --recursive https://github.com/espressif/esp-azure.git

Cloning into 'esp-azure'...
remote: Enumerating objects: 33, done.
remote: Counting objects: 100% (33/33), done.
remote: Compressing objects: 100% (27/27), done.
Receiving objects: 100% (396/396), 1.38 MiB | 0 bytes/s, done. 363

Resolving deltas: 100% (169/169), done.
Checking connectivity... done.
Submodule 'azure-iot-sdk-c' (https://github.com/Azure/azure-iot-sdk-c.git) registered for path 'azure-iot-sdk-c'
Cloning into 'azure-iot-sdk-c'...
remote: Enumerating objects: 72300, done.
remote: Total 72300 (delta 0), reused 0 (delta 0), pack-reused 72300R
Receiving objects: 100% (72300/72300), 54.88 MiB | 8.67 MiB/s, done.
Resolving deltas: 100% (48944/48944), done.
Checking connectivity... done.
Submodule path 'azure-iot-sdk-c': checked out '5d9a19a5e5ce6f4974d333bd28f432ee9284f70b'
Submodule 'azure-c-shared-utility' (https://github.com/Azure/azure-c-shared-utility.git) registered for path 'c-utility'
Submodule 'deps/parson' (https://github.com/kgabis/parson.git) registered for path 'deps/parson'
Submodule 'deps/uhttp' (https://github.com/Azure/azure-uhttp-c.git) registered for path 'deps/uhttp'
Submodule 'provisioning_client/deps/RIoT' (https://github.com/Microsoft/RIoT.git) registered for path 'provisioning_client/deps/RIoT'
Submodule 'provisioning_client/deps/utpm' (https://github.com/Azure/azure-utpm-c.git) registered for path 'provisioning_client/deps/utpm'
Submodule 'azure-uamqp-c' (https://github.com/Azure/azure-uamqp-c.git) registered for path 'uamqp'
Submodule 'azure-umqtt-c' (https://github.com/Azure/azure-umqtt-c.git) registered for path 'umqtt'
Cloning into 'c-utility'...
remote: Enumerating objects: 25, done.
remote: Counting objects: 100% (25/25), done.
remote: Compressing objects: 100% (22/22), done.
remote: Total 19875 (delta 8), reused 0 (delta 0), pack-reused 19850
Receiving objects: 100% (19875/19875), 10.27 MiB | 7.19 MiB/s, done.
Resolving deltas: 100% (13949/13949), done.
Checking connectivity... done.
Submodule path 'azure-iot-sdk-c/c-utility': checked out '6229ecb0d49b7e75fdb88d2c477e94e5b5394c43'
Submodule 'testtools/ctest' (https://github.com/Azure/azure-ctest.git) registered for path 'testtools/ctest'
Submodule 'testtools/testrunner' (https://github.com/Azure/azure-c-testrunnerswitcher.git) registered for path 'testtools/testrunner'
Submodule 'testtools/umock-c' (https://github.com/Azure/umock-c.git) registered for path 'testtools/umock-c'
Cloning into 'testtools/ctest'...
remote: Enumerating objects: 65, done.
remote: Counting objects: 100% (65/65), done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 601 (delta 27), reused 28 (delta 9), pack-reused 536
Receiving objects: 100% (601/601), 288.00 KiB | 0 bytes/s, done.
Resolving deltas: 100% (301/301), done.
Checking connectivity... done.
Submodule path 'azure-iot-sdk-c/c-utility/testtools/ctest': checked out 'ee4e06f0a5827a98417f389e013122decc1a3765'
Cloning into 'testtools/testrunner'...
remote: Enumerating objects: 56, done.
remote: Counting objects: 100% (56/56), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 386 (delta 27), reused 23 (dRelta 9), pack-reuseeceiving objects:  87% (336/d 330
Receiving objects: 100% (386/386), 898.45 KiB | 0 bytes/s, done.
Resolving deltas: 100% (203/203), done.
Checking connectivity... done.
Submodule path 'azure-iot-sdk-c/c-utility/testtools/testrunner': checked out 'c52b67126e10a2697b6d069e2b3f38b4d4229a42'
Cloning into 'testtools/umock-c'...
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (9/9), done.
emote: Total 1927 (delta 3), reused 4 (delta 1), pack-reused 1917
Receiving objects: 100% (1927/1927), 693.05 KiB | 0 bytes/s, done.
Resolving deltas: 100% (1359/1359), done.
Checking connectivity... done.
Submodule path 'azure-iot-sdk-c/c-utility/testtools/umock-c': checked out '25cc55f20fa7e6d4a830bdaa31c363236848afd9'
Submodule 'deps/azure-ctest' (https://github.com/Azure/azure-ctest.git) registered for path 'deps/ctest'
Submodule 'deps/testrunner' (https://github.com/Azure/azure-c-testrunnerswitcher.git) registered for path 'deps/testrunner'
Cloning into 'deps/ctest'...
remote: Enumerating objects: 65, done.
remote: Counting objects: 100% (65/65), done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 601 (delta 27), reused 28 (delta 9), pack-reused 536
Receiving objects: 100% (601/601), 288.00 KiB | 0 bytes/s, done.
Resolving deltas: 100% (301/301), done.
Checking connectivity... fatal: bad object 3427761cdc289eccefb88890410f1bf17bf3c4f6
fatal: remote did not send all necessary objects
fatal: clone of 'https://github.com/Azure/azure-ctest.git' into submodule path 'deps/ctest' failed
Failed to recurse into submodule path 'azure-iot-sdk-c/c-utility/testtools/umock-c'
Failed to recurse into submodule path 'azure-iot-sdk-c/c-utility'
Failed to recurse into submodule path 'azure-iot-sdk-c'
```

chinmaychhajed, 2019-11-05:
Hello @whitsonk,
This commands should have worked, but if it didn't work for you, please try this alternative method:
```
git clone https://github.com/espressif/esp-azure.git
cd esp-azure
git submodule update --init --recursive
```

shahpiyushv, 2020-06-04:
Closing this issue as a response has been provided and there are no further updates.

closed 2020-06-04 by shahpiyushv