espressif/esp-azure #69: Unable to compile SDK from components directory

GonzalezAnguita, 2019-11-17:
Hello,

I am trying to use the SDK from within a component in the **components** directory without success. I am able to run the examples successfully but when moving the azure-dependent files to *components* directory it fails.

I am getting the following error
```
CMake Error at /Users/joseto/esp/esp-mdf/esp-idf/tools/cmake/project.cmake:148 (add_subdirectory):
The binary directory

    /Users/joseto/esp/projects/azure_test/build/azure_mqtt_handler

  is already used to build a source directory.  It cannot be used to build
  source directory

    /Users/joseto/esp/projects/azure_test/components/azure_mqtt_handler

  Specify a unique binary directory name.
Call Stack (most recent call first):
  CMakeLists.txt:7 (project)

```
The **error** appears when having `#include "iothub_client.h"` in the file `azure_mqtt_handler.c`. The project structure that I have is the following.

```
azure_test
    |--> components
    |     +--> azure_mqtt_handler
    |           +--> azure_mqtt_handler.c
    |           +--> CMakeLists.txt
    |           +--> include
    |                +--> azure_mqtt_handler.h
    |--> main
    |     +--> main.c
    |     +--> CMakeLists.txt
    |     +--> Kconfig.projbuild
    |--> CMakeLists.txt
```

I have the correct **IDF_PATH** variable set and also downloaded the SDK into **ESP_PATH/esp-azure**. The following are the complete files, I've deleted everything else to try to find the issue.

**azure-test/CMakeLists.txt**
```
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set (EXTRA_COMPONENT_DIRS "${ESP_PATH}/esp-azure/port")
project(iothub_client_sample_mqtt)
```

**azure-test/main/main.c**
```
#include "azure_mqtt_handler.h"
```

**azure-test/main/CMakeLists.txt**
```
set(COMPONENT_SRCS "main.c")

set(COMPONENT_ADD_INCLUDEDIRS ".")

# requirements can't depend on config
set(COMPONENT_REQUIRES azure_mqtt_handler)

register_component()
```

**azure-test/components/azure_mqtt_handler/include/azure_mqtt_handler.h**
```
#ifndef __AZURE_MQTT_HANDLER_H__
#define __AZURE_MQTT_HANDLER_H__

#include "mdf_common.h"

#endif  // __AZURE_MQTT_HANDLER_H__
```

**azure-test/components/azure_mqtt_handler/azure_mqtt_handler.c**
```
#include "azure_mqtt_handler.h"

#include "iothub_client.h"
```

**azure-test/components/azure_mqtt_handler/CMakeLists.txt**
```
set(COMPONENT_SRCS "azure_mqtt_handler.c")
set(COMPONENT_ADD_INCLUDEDIRS "include")
set(COMPONENT_REQUIRES mcommon)

register_component()
```

Thanks

shahpiyushv, 2019-11-18:
In your file azure-test/CMakeLists.txt, can you try by inserting the keyword ENV in this line as below

set (EXTRA_COMPONENT_DIRS "$**ENV**{ESP_PATH}/esp-azure/port")

GonzalezAnguita, 2019-11-18:
Thanks for your response @shahpiyushv 

Checked the logs again and found a variable warning
```
CMake Warning (dev) at CMakeLists.txt:6 (set):
  uninitialized variable 'ESP_PATH'
This warning is for project developers.  Use -Wno-dev to suppress it.
```
Add **ENV** to the variable, fixes just the variable warning.

I tried the following and in all cases *I've got the same original issue*.

1. Hardcode the *port* absolute path.
2. Used a non existent path "/esp-azure/port_12345"
3. Removed the whole line **set (EXTRA_COMPONENT_DIRS "$ENV{ESP_PATH}/esp-azure/port")**

Also deleted the **build** directory just in case there were some cached files, nothing.

From **azure-test/CMakeLists.txt** I changed the line
```
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
```
with
```
include($ENV{MDF_PATH}/project.cmake)
```
because I was getting that the MDF library **mcommon** was not being found. Changing the previous line did not work until I moved the line **set (EXTRA_COMPONENT_DIRS "$ENV{ESP_AZURE}/port")** to the file **azure-test/components/azure_mqtt_handler/CMakeLists.txt** and the error message changed to the following
```
Checking Python dependencies...
Python requirements from /Users/joseto/esp/esp-mdf/esp-idf/requirements.txt are satisfied.
Running ninja in directory /Users/joseto/esp/projects/iothub_client_sample_mqtt/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building empty aws_iot component due to configuration
-- Building empty aliyun iot sdk component due to configuration
-- Building empty aws_iot component due to configuration
-- Component names: soc log heap xtensa-debug-module app_trace freertos vfs newlib esp_ringbuf driver esp_event ethernet lwip tcpip_adapter partition_table app_update spi_flash mbedtls micro-ecc bootloader_support nvs_flash pthread smartconfig_ack wpa_supplicant esp32 cxx nghttp esp-tls tcp_transport esp_http_client json mcommon azure_mqtt_handler main esp-aliyun miniz mwifi mespnow console mdns esp_http_server mupgrade bt mconfig mlink maliyun_linkkit wear_levelling sdmmc fatfs spiffs mdebug asio jsmn aws_iot bootloader coap esp_adc_cal esp_https_ota esptool_py expat freemodbus idf_test libsodium mqtt openssl protobuf-c protocomm ulp wifi_provisioning
-- Component paths: /Users/joseto/esp/esp-mdf/esp-idf/components/soc;/Users/joseto/esp/esp-mdf/esp-idf/components/log;/Users/joseto/esp/esp-mdf/esp-idf/components/heap;/Users/joseto/esp/esp-mdf/esp-idf/components/xtensa-debug-module;/Users/joseto/esp/esp-mdf/esp-idf/components/app_trace;/Users/joseto/esp/esp-mdf/esp-idf/components/freertos;/Users/joseto/esp/esp-mdf/esp-idf/components/vfs;/Users/joseto/esp/esp-mdf/esp-idf/components/newlib;/Users/joseto/esp/esp-mdf/esp-idf/components/esp_ringbuf;/Users/joseto/esp/esp-mdf/esp-idf/components/driver;/Users/joseto/esp/esp-mdf/esp-idf/components/esp_event;/Users/joseto/esp/esp-mdf/esp-idf/components/ethernet;/Users/joseto/esp/esp-mdf/esp-idf/components/lwip;/Users/joseto/esp/esp-mdf/esp-idf/components/tcpip_adapter;/Users/joseto/esp/esp-mdf/esp-idf/components/partition_table;/Users/joseto/esp/esp-mdf/esp-idf/components/app_update;/Users/joseto/esp/esp-mdf/esp-idf/components/spi_flash;/Users/joseto/esp/esp-mdf/esp-idf/components/mbedtls;/Users/joseto/esp/esp-mdf/esp-idf/components/micro-ecc;/Users/joseto/esp/esp-mdf/esp-idf/components/bootloader_support;/Users/joseto/esp/esp-mdf/esp-idf/components/nvs_flash;/Users/joseto/esp/esp-mdf/esp-idf/components/pthread;/Users/joseto/esp/esp-mdf/esp-idf/components/smartconfig_ack;/Users/joseto/esp/esp-mdf/esp-idf/components/wpa_supplicant;/Users/joseto/esp/esp-mdf/esp-idf/components/esp32;/Users/joseto/esp/esp-mdf/esp-idf/components/cxx;/Users/joseto/esp/esp-mdf/esp-idf/components/nghttp;/Users/joseto/esp/esp-mdf/esp-idf/components/esp-tls;/Users/joseto/esp/esp-mdf/esp-idf/components/tcp_transport;/Users/joseto/esp/esp-mdf/esp-idf/components/esp_http_client;/Users/joseto/esp/esp-mdf/esp-idf/components/json;/Users/joseto/esp/esp-mdf/components/mcommon;/Users/joseto/esp/projects/iothub_client_sample_mqtt/components/azure_mqtt_handler;/Users/joseto/esp/projects/iothub_client_sample_mqtt/main;/Users/joseto/esp/esp-mdf/components/third_party/esp-aliyun;/Users/joseto/esp/esp-mdf/components/third_party/miniz;/Users/joseto/esp/esp-mdf/components/mwifi;/Users/joseto/esp/esp-mdf/components/mespnow;/Users/joseto/esp/esp-mdf/esp-idf/components/console;/Users/joseto/esp/esp-mdf/esp-idf/components/mdns;/Users/joseto/esp/esp-mdf/esp-idf/components/esp_http_server;/Users/joseto/esp/esp-mdf/components/mupgrade;/Users/joseto/esp/esp-mdf/esp-idf/components/bt;/Users/joseto/esp/esp-mdf/components/mconfig;/Users/joseto/esp/esp-mdf/components/mlink;/Users/joseto/esp/esp-mdf/components/maliyun_linkkit;/Users/joseto/esp/esp-mdf/esp-idf/components/wear_levelling;/Users/joseto/esp/esp-mdf/esp-idf/components/sdmmc;/Users/joseto/esp/esp-mdf/esp-idf/components/fatfs;/Users/joseto/esp/esp-mdf/esp-idf/components/spiffs;/Users/joseto/esp/esp-mdf/components/mdebug;/Users/joseto/esp/esp-mdf/esp-idf/components/asio;/Users/joseto/esp/esp-mdf/esp-idf/components/jsmn;/Users/joseto/esp/esp-mdf/esp-idf/components/aws_iot;/Users/joseto/esp/esp-mdf/esp-idf/components/bootloader;/Users/joseto/esp/esp-mdf/esp-idf/components/coap;/Users/joseto/esp/esp-mdf/esp-idf/components/esp_adc_cal;/Users/joseto/esp/esp-mdf/esp-idf/components/esp_https_ota;/Users/joseto/esp/esp-mdf/esp-idf/components/esptool_py;/Users/joseto/esp/esp-mdf/esp-idf/components/expat;/Users/joseto/ [... cut ...]
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

I can see two lines standing out here
```
[196/745] Building C object azure_mqtt_handler/CMakeFiles/azure_mqtt_handler.dir/azure_mqtt_handler.c.obj
( . . . )
../components/azure_mqtt_handler/azure_mqtt_handler.c:4:27: fatal error: iothub_client.h: No such file or directory
```

and

```
[197/745] Building C object main/CMakeFiles/main.dir/main.c.obj
( . . . )
../main/main.c:1:32: fatal error: azure_mqtt_handler.h: No such file or directory
```

shahpiyushv, 2019-11-18:
@GonzalezAnguita , this may sound naive, but have you indeed set the ESP_PATH using `export ESP_PATH` command so that it points to the PATH you mentioned where you have cloned esp-azure (ESP_PATH/esp-azure)? Ideally, this single change should have been enough.

GonzalezAnguita, 2019-11-18:
@shahpiyushv yes, all ENV variables are correctly set. It’s weird since I am using the same in the examples and those work perfectly. The difference is in the component folder structure

tinfoilbroker, 2019-11-18:
I'm experiencing something similar to what @GonzalezAnguita is experiencing, but with an even simpler project structure.

```
esp-project
    |--> main
    |     +--> main.c
    |     +--> CMakeLists.txt
    |--> CMakeLists.txt
    |--> sdkconfig
```
**esp-project/CMakeLists.txt**
```
cmake_minimum_required(VERSION 3.5)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(EXTRA_COMPONENT_DIRS "${PROJECT_DIR}/.../esp-azure/port")
project(esp-project)
```
**project/main/CMakeLists.txt**
```
idf_component_register(
	SRCS "main.c"
        INCLUDE_DIRS ""
)
```
**project/main/main.c**
```
#include "iothub.h"
#include "stdio.h"

void app_main(void)
{
	printf("Hello World.\n");
}
```

When compiling the following error message occurs:
```
../main/main.c:1:10: fatal error: iothub.h: No such file or directory
 #include "iothub.h"
          ^~~~~~~~~~
```
The project compiles fine when I remove `#include "iothub.h"` from main.c. Changing `set(EXTRA_COMPONENT_DIRS "${PROJECT_DIR}/.../esp-azure/port")` or removing it doesn't do anything as well.

GonzalezAnguita, 2019-11-19:
After some experimenting, I realized some issues, the first one is that if you import the azure sdk as this
```
set (EXTRA_COMPONENT_DIRS "$ENV{ESP_PATH}/esp-azure/port")
```
it will cause issues with the project `components` directory so you must use
```
set (EXTRA_COMPONENT_DIRS "${EXTRA_COMPONENT_DIRS} $ENV{ESP_PATH}/esp-azure/port")
```

Second one is that it seems that you can't import the SDK using the previous line from inside a `COMPONENT_DIR` trying to do so, will result in errors like `No such file or directory`. **Also** the import doesn't propagate into the `components` directory so you can't import it from the `project/CMakeLists.txt` file trying to use it in a component.

# WORKAROUND
`Disclaimer: is not the cleanest solution but works`
So how to be able to use the SDK and also custom components?
1. Set under project/CMakeLists.txt the line
```
set (EXTRA_COMPONENT_DIRS "${EXTRA_COMPONENT_DIRS} $ENV{ESP_PATH}/esp-azure/port")
```
2. Create a directory within the `main` directory called `azure_handler` for example. And inside place `azure_handler.h` and `azure_handler.c`.
3. The file `project/main/CMakeLists.txt` should be something like this
```
set(COMPONENT_ADD_INCLUDEDIRS "azure_handler")

set(COMPONENT_SRCS
    "azure_handler/azure_handler.c"
    "main.c")

register_component()
```

Hope this helps someone.
JT

PS: If you have a better and cleanest solution please take the time to share it. Thanks

shahpiyushv, 2019-11-19:
@GonzalezAnguita , keeping rest everything the same as you already had i.e. esp-azure under ESP_PATH and this structure

```
azure_test
    |--> components
    |     +--> azure_mqtt_handler
    |           +--> azure_mqtt_handler.c
    |           +--> CMakeLists.txt
    |           +--> include
    |                +--> azure_mqtt_handler.h
    |--> main
    |     +--> main.c
    |     +--> CMakeLists.txt
    |     +--> Kconfig.projbuild
    |--> CMakeLists.txt
```
can you try setting extra components dirs as below and check?

```
set (EXTRA_COMPONENT_DIRS "$ENV{ESP_PATH}/port" "${CMAKE_CURRENT_BINARY_DIR}/components")
```

GonzalezAnguita, 2019-11-23:
@shahpiyushv tried that and the component complains
```
../components/azure_mqtt_handler/azure_mqtt_handler.c:4:27: fatal error: iothub_client.h: No such file or directory
```

liudr, 2019-11-25:
I'm not an advanced user of this sdk but are we looking at an issue where compiler can't find header files? In Makefile I have this:

`# EXTRA_COMPONENT_DIRS += $(PROJECT_PATH)/../../../esp-azure

EXTRA_COMPONENT_DIRS += $(AZURE_PATH)
`
I never copied any components from azure to my project folder. Instead, I defined an additional env AZURE_PATH. It is absolute. The above shows what I changed.

GonzalezAnguita, 2020-01-25:
I just found the issue to this problem. It was not the component but a dependent component in

**azure-test/components/azure_mqtt_handler/CMakeLists.txt**
```
set(COMPONENT_SRCS "azure_mqtt_handler.c")
set(COMPONENT_ADD_INCLUDEDIRS "include")
set(COMPONENT_REQUIRES  port mcommon)

register_component()
```
The dependency **port** was missing under **COMPONENT_REQUIRES**

Also keep in mind the file
**azure-test/CMakeLists.txt**
```
include($ENV{MDF_PATH}/project.cmake)
set(EXTRA_COMPONENT_DIRS "${EXTRA_COMPONENT_DIRS} $ENV{ESP_AZURE}/port")
project(measure_mesh)
```
to have it include **$ENV{ESP_AZURE}/port**.

Thanks for your time!

closed 2020-01-25 by GonzalezAnguita