espressif/esp-azure #70: error: 'request_id' may be used uninitialized

fromeijn, 2019-11-19:
While compiling with `Release (-Os)` compiler complains about `'request_id' may be used uninitialized`, and i'm unable to build with Release. `Debug (-Og)` works tho.
```
CC build/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.o
/home/floris/Documents/git/hcb-fw/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c: In function 'mqtt_notification_callback':
/home/floris/Documents/git/hcb-fw/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c:1518:32: error: 'request_id' may be used uninitialized in this function [-Werror=maybe-uninitialized]
                             if (request_id == msg_entry->packet_id)
                                ^
cc1: some warnings being treated as errors
make[1]: *** [/home/floris/Documents/esp/esp-idf/make/component_wrapper.mk:290: azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.o] Error 1
make: *** [/home/floris/Documents/esp/esp-idf/make/project.mk:552: component-esp-azure-build] Error 2
```
ESP-IDF V3.3 [0a0f2ca](https://github.com/espressif/esp-idf/commit/0a0f2caa1d30e0f8cbfe31475093492e09995763)
ESP-Azure [6f054a8](https://github.com/espressif/esp-azure/commit/6f054a865f8cb58670cdca3185060a7a16538d24)

shahpiyushv, 2019-11-20:
@fromeijn , can you try initialising the value of request id as 0 (size_t request_id = 0;) on this line: https://github.com/Azure/azure-iot-sdk-c/blob/5d9a19a5e5ce6f4974d333bd28f432ee9284f70b/iothub_client/src/iothubtransport_mqtt_common.c#L1496

fromeijn, 2019-11-26:
yes after applying this it works:
```
diff --git a/iothub_client/src/iothubtransport_mqtt_common.c b/iothub_client/src/iothubtransport_mqtt_common.c
index 6b5a3ce49..1b4366470 100644
--- a/iothub_client/src/iothubtransport_mqtt_common.c
+++ b/iothub_client/src/iothubtransport_mqtt_common.c
@@ -1493,7 +1493,7 @@ static void mqtt_notification_callback(MQTT_MESSAGE_HANDLE msgHandle, void* call
             IOTHUB_IDENTITY_TYPE type = retrieve_topic_type(topic_resp, STRING_c_str(transportData->topic_InputQueue));
             if (type == IOTHUB_TYPE_DEVICE_TWIN)
             {
-                size_t request_id;
+                size_t request_id = 0;
                 int status_code;
                 bool notification_msg;
                 if (parse_device_twin_topic_info(topic_resp, &notification_msg, &request_id, &status_code) != 0)
diff --git a/serializer/src/agenttypesystem.c b/serializer/src/agenttypesystem.c
index 06e46e193..c74c5d4c1 100644
--- a/serializer/src/agenttypesystem.c
+++ b/serializer/src/agenttypesystem.c
@@ -3280,7 +3280,7 @@ AGENT_DATA_TYPES_RESULT CreateAgentDataType_From_String(const char* source, AGEN
             {
                 int32_t int32Value;
                 unsigned char isNegative;
-                uint32_t uint32Value;
+                uint32_t uint32Value = 0;
                 const char* pos;
                 size_t strLength;
```

fromeijn, 2019-12-30:
made a quick fix https://github.com/fromeijn/esp-azure

Lisa999, 2020-10-30:
@shahpiyushv : will this be fixed in a future release?

shahpiyushv, 2020-11-02:
@Lisa999 , the relevant code was from upstream azure-iot-sdk-c and not esp-azure. We haven't explicitly submitted any fix, but have updated azure-iot-sdk-c to a newer LTS release.