espressif/esp-azure #71: Is blob uploaded working?

liudr, 2019-11-25:
Since there isn't any example on blob upload, I'm wondering if this feature is available in this port.

More specifically, is either function actually working?

IoTHubDeviceClient_LL_UploadToBlob()
IoTHubDeviceClient_UploadToBlobAsync()

Is this thing defined or where should I find it if it is defined:

#ifndef DONT_USE_UPLOADTOBLOB

I found their definitions in the source code. What is the LL? Lower level? What is the difference? The non_LL has a callback and a pointer probably for the callback's parameters. Is that the difference? The non_LL is asyn because it creates a task and hands off the work to the task? Any pointer to the main azure iot port doc is greatly appreciated.
A related issue: [https://github.com/Azure/azure-iot-sdk-c/issues/1343]

I have trouble linking my code once I added IoTHubDeviceClient_LL_UploadToBlob(). That related issue has a solution but I tried to add those components in components.mk but got "missing separator" error. Need some support please!

B-Temia, 2019-11-25:
Hi,

The blob upload sample from MS works fine with sas authentication but not with certificate. I have a issue open about that. I will provide more infos to espressif to receive more help. The problem come from tlsio_esp_tls.c and not from IoT Hub sdk directly.

[https://github.com/espressif/esp-azure/issues/65](url)

liudr, 2019-11-26:
> Hi,
>
> The blob upload sample from MS works fine with sas authentication but not with certificate. I have a issue open about that. I will provide more infos to espressif to receive more help. The [...]

I have trouble compiling my code once I added IoTHubDeviceClient_LL_UploadToBlob(). Could you share how you modified your project to compile? Thanks.

liudr, 2019-11-29:
Any support? Only 3 ported example programs sounds like very little support!

liudr, 2019-12-01:
What about a response "we WILL look into it soon"? I am stuck trying to upload blob. It could be a make-or-break situation, meaning a huge quantity potential purchase of ESP32 processors or not. Please provide some support.

frederic-thibault, 2019-12-02:
@liudr Did you make it build?  If not look at this issue I have indicate what is needed to build it.

https://github.com/espressif/esp-azure/issues/65

I don't have a lot of support for the upload since the guy has no idea to where to start and I didn't take the time to give him some info about that. 

On my side I started to work on a plan B and upload the file to an app service by reusing the certificate for the authorization. Since .netcore 3 support it, it is a viable solution if they can't fix the problem.

https://damienbod.com/2019/06/13/certificate-authentication-in-asp-net-core-3-0/

liudr, 2019-12-02:
federic thanks! Did you define ${AZURE_IOT_SDK} to be the absolute path ot your esp-azure folder?
I'm using a PC by the way. I'm also using make. It seems you're using Cmake. Maybe I should move over to cmake. What do you think?

B-Temia, 2019-12-02:
@liudr 
Hi,

Instead of using 
set (EXTRA_COMPONENT_DIRS "${CMAKE_CURRENT_BINARY_DIR}/../../../port") 
like defined in the example, I created an environment variable ESP_AZURE=D:/ESP/esp-azure (you have to use / and not \, I am also on PC).

set (EXTRA_COMPONENT_DIRS "$ENV{ESP_AZURE}/port")

You dont have to touch ${AZURE_IOT_SDK} it's define in the CMakeList. based on the current folder.