espressif/esp-azure #77: About SNTP

liudr, 2020-01-06:
Is the call to IoTHubDeviceClient_LL_CreateFromConnectionString(connectionString, protocol) starting SNTP or where in the code does SNTP get initialized? I wish to use time but don't know how and where SNTP is started. I suspect it's inside azure client code, is it? Does azure code periodically sync time by calling some SNTP functions?

Darkenkade42, 2020-01-08:
https://github.com/espressif/esp-azure/blob/4a6149ffa40955175abd705057ccf521df7ec658/port/src/platform_esp.c#L24

https://github.com/espressif/esp-azure/blob/4a6149ffa40955175abd705057ccf521df7ec658/port/src/agenttime_esp.c#L44

liudr, 2020-01-09:
Thanks for pointing out in the source code. I'm new to sntp and frankly am a bit lost in the source code you pointed out. It seems to me that the sntp_get_current_timestamp() function just calls obtain_time() which runs an infinite loop until year is > 2016 (when esp32 was "born"?). This sounds like a good way for obtaining time from having no correct time after boot. Indeed the sntp sample code in esp-idf has the exactly name logic if not identical code.

What about updating time with time server? My ESP32 gains about 15 minutes every 12 hours if I just call initialize_sntp() once like in azure. I know I shouldn't call it more than once. I found a version of  initialize_sntp() in sntp example code that has these lines. Focusing on line 2:

    ESP_LOGI(TAG, "Initializing SNTP");
    sntp_setoperatingmode(SNTP_OPMODE_POLL);
    sntp_setservername(0, "pool.ntp.org");
    sntp_init();

So am I supposed to call some poll function each time I wish to re-sync? Is there an auto mode that I can set re-sync period for such as 5 minutes within which esp32 rtc time slip isn't too bad?

Thanks again!

Darkenkade42, 2020-01-09:
The SNTP_OPMODE_POLL option sets automatic polling, which by default polls every hour. Depending on the ESP-IDF version you're using, you could use this new API to set sync interval:

https://github.com/espressif/esp-idf/blob/ca8fac876ff8deb23fcaf37bd7ce9127f8c6c438/components/lwip/include/apps/sntp/sntp.h#L133 

15 min every 12 hours sounds rather extreme, you may wish to enable SNTP debug logs if the issue persists. There also might be an issue with RTC itself, which is beyond the scope of my knowledge.

liudr, 2020-01-15:
After some checking with my azure program and a WROVER-B board I made, I see that the RTC is within seconds of actual time so it's satisfactory. I wonder what was wrong with the esp-idf sntp example code or with the WROOOM I used for testing.

I see that the feature is not included in the latest release, 4.0. I am on ESP-IDF V3.3 and will consider migrating to 4.x at a later time when there are enough benefits against the hassle. What time interval do you recommend if I DO migrate to 4.x and gain the capability to set period? Thank you!

shahpiyushv, 2020-06-04:
@Darkenkade42 thank you for helping out here. @liudr , seems that your issue has now been resolved. Just choosing SNTP_OPMODE_POLL should be sufficient. If it does not work, please feel free to reopen this issue.

closed 2020-06-04 by shahpiyushv