espressif/esp-azure #80: Using Direct Methods with great amount of data, the wifi crashes

sfarras, 2020-01-27:
When we use Direct Methods with payload bigger than 16kbytes~20Kbytes , some part of the wifi crashes and doesn't recover until reset.  The amount of data to reproduce the issue can vary slightly, depending of the free RAM. Always crashes when json size is above 20kbytes. 

The wifi module shows this message :
W (100500) wifi: alloc eb len=24 type=3 fail, heap:4024328
W (100500) wifi: m f null

And this message is repeated indefinitely.. 
Info: Error from SSL_write: -26752
Info: Error from SSL_write: -26752
Info: Error from SSL_write: -26752
...

We tried a lot of menuconfig combinations in aim to free as much RAM as possible. The result is more or less the same. The max amount of data to be sent (or receive) without problem can vary from 10kb to 20kb. Never more than 20 kbytes.
The hardware used is an ESP32-WROVER module with 4MBytes of external RAM.
We develop on esp-idf v3.3.1 framework version

AZURE IoT HUB can manage up to 128kbytes as explained in https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-devguide-direct-methods , and the WROVER module has enough RAM to manage these messages.

shahpiyushv, 2020-01-27:
@sfarras , can you share your sdkconfig file?

sfarras, 2020-01-27:
This is my sdkconfig

Thank you!

Missatge de shahpiyushv <notifications@github.com> del dia dl., 27 de gen.
2020 a les 15:36:

sfarras, 2020-01-27:
This is my sdkconfig 

#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#
CONFIG_IDF_TARGET="esp32"

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
CONFIG_APP_EXCLUDE_PROJECT_VER_VAR=
CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR=

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V=
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=
CONFIG_BOOTLOADER_WDT_ENABLE=y
CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE=
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
CONFIG_APP_ROLLBACK_ENABLE=

#
# Security features
#
CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT=
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
CONFIG_ESPTOOLPY_BAUD_115200B=
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD_2MB=y
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=2000000
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=y
CONFIG_FLASHMODE_DOUT=
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_80M=
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
CONFIG_ESPTOOLPY_FLASHFREQ_26M=
CONFIG_ESPTOOLPY_FLASHFREQ_20M=
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
[... 731 lines cut ...]
# Unity unit testing library
#
CONFIG_UNITY_ENABLE_FLOAT=y
CONFIG_UNITY_ENABLE_DOUBLE=y
CONFIG_UNITY_ENABLE_COLOR=
CONFIG_UNITY_ENABLE_IDF_TEST_RUNNER=y
CONFIG_UNITY_ENABLE_FIXTURE=

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096

#
# Wi-Fi Provisioning Manager
#
CONFIG_WIFI_PROV_SCAN_MAX_ENTRIES=16

sfarras, 2020-01-31:
Hi,
I think esp-azure has no issues on this.
The problem is how I manage the internal RAM.
 My work around was setting CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL to very small value, leaving available a big amount of internal RAM to wifi and ssl layers....

closed 2020-02-20 by sfarras