espressif/esp-azure #83: sending multiple messages to azure causes esp reboot

enerdata, 2020-02-20:
Hi there,

i'm sending multiple messages to azure ,with message to message retention delay of 1 sec , up to 10 or 20 messages it is ok.

after that messages causing system error and rebooting itself..

see the below error 

Thanks in advance !!!

onfirmation[1] rDeceived for message tracking id = 0 with result = IOTHUB_CLIENT_CONFIRMATION_OK
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400f54ec  PS      : 0x00060130  A0      : 0x800f8862  A1      : 0x3ffd7410  
0x400f54ec: Map_Destroy at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/c-utility/src/map.c:48 (discriminator 1)

A2      : 0xfefefefe  A3      : 0x00000000  A4      : 0x3ffd7344  A5      : 0x00000000  
A6      : 0x0000001d  A7      : 0xff000000  A8      : 0x800829ec  A9      : 0x3ffd70c0  
A10     : 0x3ffae9e4  A11     : 0x00000004  A12     : 0x3ffd7344  A13     : 0x3ffae98c  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfefeff06  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff8  

ELF file SHA256: 3fd942adcc7c5acc1d222086862cd2ea1bc48be49e927ec65b36db0551810fb3

Backtrace: 0x400f54e9:0x3ffd7410 0x400f885f:0x3ffd7430 0x400f942b:0x3ffd7450 0x400d4e28:0x3ffd7470 0x400f6373:0x3ffd7490 0x400f988a:0x3ffd74b0 0x400f98d1:0x3ffd74e0 0x400fd9df:0x3ffd7510 0x400fe32e:0x3ffd7540 0x400fef81:0x3ffd7560 0x400fd34d:0x3ffd7580 0x40180e81:0x3ffd75a0 0x400f48bb:0x3ffd7600 0x40181467:0x3ffd7620 0x400fe227:0x3ffd7640 0x400fc7c2:0x3ffd7670 0x400f9495:0x3ffd76a0 0x400f68e9:0x3ffd76c0 0x400f5e01:0x3ffd76f0 0x400d5070:0x3ffd7710 0x400d4456:0x3ffdc760 0x40093d49:0x3ffdc780
0x400f54e9: Map_Destroy at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/c-utility/src/map.c:48

0x400f885f: DestroyMessageData at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothub_message.c:91

0x400f942b: IoTHubMessage_Destroy at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothub_message.c:1025

0x400d4e28: SendConfirmationCallback at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/examples/Rightdeal_GateWay_rev_00_APSTASER_CONSTR/main/iothub_client_sample_mqtt.c:147

0x400f6373: IoTHubClientCore_LL_SendComplete at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothub_client_core_ll.c:510

0x400f988a: sendMsgComplete at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c:589

0x400f98d1: mqtt_operation_complete_callback at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c:1680

0x400fd9df: recvCompleteCallback at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/umqtt/src/mqtt_client.c:786

0x400fe32e: completePacketData at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/umqtt/src/mqtt_codec.c:601

0x400fef81: mqtt_codec_bytesReceived at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/umqtt/src/mqtt_codec.c:1120

0x400fd34d: onBytesReceived at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/umqtt/src/mqtt_client.c:441

0x40180e81: dowork_read at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/port/src/tlsio_esp_tls.c:410

0x400f48bb: tlsio_esp_tls_dowork at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/port/src/tlsio_esp_tls.c:496

0x40181467: xio_dowork at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/c-utility/src/xio.c:165

0x400fe227: mqtt_client_dowork at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/umqtt/src/mqtt_client.c:1247

0x400fc7c2: IoTHubTransport_MQTT_Common_DoWork at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c:3166

0x400f9495: IoTHubTransportMqtt_DoWork at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransportmqtt.c:121

0x400f68e9: IoTHubClientCore_LL_DoWork at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothub_client_core_ll.c:2070

0x400f5e01: IoTHubClient_LL_DoWork at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/azure-iot-sdk-c/iothub_client/src/iothub_client_ll.c:91

0x400d5070: iothub_client_sample_mqtt_run at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/examples/Rightdeal_GateWay_rev_00_APSTASER_CONSTR/main/iothub_client_sample_mqtt.c:269

0x400d4456: RS_azure_run at /home/saiteja_vdeal/esp/esp-idf/components/esp-azure/examples/Rightdeal_GateWay_rev_00_APSTASER_CONSTR/main/azure_main.c:554

0x40093d49: vPortTaskWrapper at /home/saiteja_vdeal/esp/esp-idf/components/freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57

enerdata, 2020-02-20:
![Uploading Screenshot from 2020-02-20 16-08-24.png
please find the above attachment

closed 2020-02-25 by enerdata

pw007, 2020-08-09:
So, Enerdata, how you solve this problem?