espressif/esp-azure #84: Error al inicializar SNTP con esp32 y Azure iot

emgm, 2020-03-12:
Estoy configurando un microcontrolador ESP32 para conectarlo con Azure iot hub, cuando establezco que publique cada 3 minutos estoy presentando perdida de conexion y me presenta el siguiente error.

Error: Time:Thu Mar 12 00:28:54 2020 File:C:\Users\egonzalez\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\AzureIoT\src\Esp32MQTTClient.cpp Func:SendEventOnce Line:316 Waiting for send confirmation, time is up 10011

Ya he revisado los foros 
https://github.com/VSChina/ESP32_AzureIoT_Arduino/issues/5

https://github.com/ustccw/AzureESP32/issues/4

Pero no he podido resolver satisfactoriamente.

De antemano agradezco la atención prestada y quedando atento.

shahpiyushv, 2020-06-04:
@emgm , seems you aren't using esp-azure in your code, but the arduino variant. ESP32_AzureIoT_Arduino isn't maintained by Espressif.

emgm, 2020-06-05:
Si, dos cuestiones.

1) Tiene alguna documentación de buenas practicas, en la que me pueda sugerir para conectar adecuadamente esp32 con azure iot

2) Cuando conecto esp32 con azure iot, me genera error si configuro el envio de mensajes cada 3 minutos. La conexión se restablece