espressif/esp-azure #85: Broken submodule links in the preview/pnp_example branch

mingqxu7, 2020-04-04:
% git clone --recursive -b preview/pnp_example https://github.com/espressif/esp-azure.git
...
...
remote: Compressing objects: 100% (47/47), done.
remote: Total 495 (delta 39), reused 51 (delta 22), pack-reused 416
Receiving objects: 100% (495/495), 301.04 KiB | 392.00 KiB/s, done.
Resolving deltas: 100% (295/295), done.
Submodule path 'azure-iot-sdk-c/umqtt/deps/umock-c/deps/testrunner/deps/azure-macro-utils-c': checked out '7523af934fc4d9423111e358f49b19314ec9c3e3'
Failed to recurse into submodule path 'azure-iot-sdk-c/provisioning_client/deps/utpm'
Failed to recurse into submodule path 'azure-iot-sdk-c/uamqp'
Failed to recurse into submodule path 'azure-iot-sdk-c'

shahpiyushv, 2020-04-16:
I just tried this out myself on MacOS, just copy-pasting your command `git clone --recursive -b preview/pnp_example https://github.com/espressif/esp-azure.git` and it worked for me. Can you double check?

shahpiyushv, 2020-07-02:
Closing this issue as there hasn't been any activity on it for several days.

closed 2020-07-02 by shahpiyushv